tinygo-org/tinygo #3013: USB CDC not flashing over old tinygo binaries

kenbell, 2022-07-22:
Scenario:
1. Flash Pico with an 'old' tinygo binary that doesn't have USB support
2. Upgrade tinygo to latest dev
3. Run `tinygo flash -target pico ./src/examples/serial/`
4. Put Pico into flash mode (hold bootsel, plug-in)

Expected:
Flash works

Actual:
`error: unable to locate a serial port`

Recovery:
Edit 

Thoughts:
1. We shouldn't require editing of rp2040.json to recover an RP2040 target
2. We shouldn't break the old workflow of putting pico into flash mode using bootsel

Suggested fix:
`tinygo flash` should check to see if `RPI-RP2` volume exists before trying the serial port magic.

sago35, 2022-07-22:
@kenbell 
Perhaps this will fix it.
We could check the MSD first, but in most cases it is more efficient to reset the port first.
https://github.com/tinygo-org/tinygo/pull/2992

kenbell, 2022-07-22:
Thanks @sago35 - that change fixes this issue

closed 2022-08-04 by deadprogram