tinygo-org/tinygo #3019: Ranging over a large array makes the compiler eat my ram

Jille, 2022-07-25:
```
// tinygo build -o repro.wasm -target wasi repro.go

package main

func main() {
	var buckets [1 << 24]int16
	for _, b := range buckets {
		_ = b
	}
}
```

`tinygo` starts consuming all my RAM.

Stacktrace of the compiler:

```
goroutine 75 [syscall]:
tinygo.org/x/go-llvm._Cfunc_LLVMRunFunctionPassManager(0x60000021c3a0, 0x600002c04208)
	_cgo_gotypes.go:8863 +0x4c
tinygo.org/x/go-llvm.PassManager.RunFunc.func1({0x4225af9?}, {0x2?})
	/Users/runner/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20220420140351-512c94c1e71f/ir.go:2010 +0x70
tinygo.org/x/go-llvm.PassManager.RunFunc({0x600003500580?}, {0xc000143cf0?})
	/Users/runner/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20220420140351-512c94c1e71f/ir.go:2010 +0x19
github.com/tinygo-org/tinygo/builder.Build.func1(0x0)
	/Users/runner/work/tinygo/tinygo/builder/build.go:393 +0xc2b
github.com/tinygo-org/tinygo/builder.runJob(0xc00010f1a0, 0x0?)
	/Users/runner/work/tinygo/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/runner/work/tinygo/tinygo/builder/jobs.go:123 +0x5db
```

```
tinygo version 0.23.0 darwin/amd64 (using go version go1.18 and LLVM version 14.0.0)
Darwin 21.3.0 Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_ARM64_T6000 arm64
```

dgryski, 2022-08-09:
Ranging over an addressable array as you are doing here causes Go to have to copy the array.  You can prevent this by instead ranging over a slice:
```
	for _, b := range buckets[:] {
```

This is required by the specification.  Upstream Go also has this "problem" but it is effectively invisible due to the better garbage collector.

deadprogram, 2022-08-09:
Thank you for the excellent explanation @dgryski 

Now closing as answered, please reopen if needed.

closed 2022-08-09 by deadprogram

Jille, 2022-08-09:
Agreed that ranging over the slice is the easy fix.

However, I'd still argue that the *compiler* taking 30GB+ RAM is a bug.

aykevl, 2022-08-09:
I agree this is a bug and should be fixed.
When we looked at the problem it appears that a LLVM pass is going wrong, it somehow hangs and eats all RAM.

dgryski, 2022-08-09:
Good point.  Seems like there must be something about how we're generating the IR for this particular case that causes the memory bloat.

dgryski, 2024-10-03:
Sadly still happening.  Interesting to note that
```
package main

func main() {
	var buckets [1 << 24]int16
	var total int16
	for _, b := range buckets {
		total += b
	}
	_ = total
	//println(total)
}
```

without the `println` it's fine, but with the `println` (probably because the llvm dead-code pass removes it all) but *with* the `println` we get stuck in optimizer pass hell.

```
~/go/src/github.com/dgryski/bug/buf $ time tinygo run main.go
^\SIGQUIT: quit
PC=0x1170d13ca m=0 sigcode=0
signal arrived during cgo execution

goroutine 130 gp=0xc000103500 m=0 mp=0x10e7308a0 [syscall]:
runtime.cgocall(0x10e3011e0, 0xc00153fa40)
	/Users/dgryski/go/src/go.googlesource.com/go/src/runtime/cgocall.go:167 +0x4b fp=0xc00153fa18 sp=0xc00153f9e0 pc=0x10df3e18b
tinygo.org/x/go-llvm._Cfunc_LLVMRunPasses(0x7fd916f04b50, 0x6000014b2020, 0x0, 0x6000014c9100)
	_cgo_gotypes.go:7825 +0x4c fp=0xc00153fa40 sp=0xc00153fa18 pc=0x10e14754c
tinygo.org/x/go-llvm.Module.RunPasses.func2({0xc00140e690?}, 0x6000014b2020, {0x10e2418c5?}, {0x10e452bc0?})
	/Users/dgryski/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20240627184919-3b50c76783a8/passes.go:43 +0xb1 fp=0xc00153faa8 sp=0xc00153fa40 pc=0x10e159571
tinygo.org/x/go-llvm.Module.RunPasses({0x10e3131c4?}, {0xc00140e690?, 0xc0015e1d60?}, {0x1?}, {0x1?})
	/Users/dgryski/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20240627184919-3b50c76783a8/passes.go:43 +0x85 fp=0xc00153fb38 sp=0xc00153faa8 pc=0x10e159305
github.com/tinygo-org/tinygo/transform.Optimize({0x10e2933bb?}, 0xc0001ccc00)
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/transform/optimizer.go:152 +0xc1f fp=0xc00153fdc8 sp=0xc00153fb38 pc=0x10e276eff
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x60000169c000?}, 0xc0001ccc00, 0xc0001df650)
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/builder/build.go:1133 +0x9d fp=0xc00153fe10 sp=0xc00153fdc8 pc=0x10e29833d
github.com/tinygo-org/tinygo/builder.Build.func5(0x0?)
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/builder/build.go:582 +0x61f fp=0xc00153ff80 sp=0xc00153fe10 pc=0x10e292f5f

```