tinygo-org/tinygo #3020: Casting an go:embed string to a []byte crashes the compiler

Jille, 2022-07-26:
```
$ tinygo run repro.go      
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x4068580]

goroutine 32 [running]:
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0003d6fa0, 0xc0004c15e0, {0xc000fe1810, 0x4, 0x0?}, 0xc002c7ab40, {0xc000ff3a20, 0xc})
	/Users/runner/work/tinygo/tinygo/interp/interpreter.go:375 +0x5548
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0003d6fa0, 0xc000fec280, {0xc0033e4b50, 0x3, 0x0?}, 0xc002c7a510, {0xc000ff38d0, 0x8})
	/Users/runner/work/tinygo/tinygo/interp/interpreter.go:544 +0x84b4
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc0003d6fa0, 0xc000fec230, {0x0, 0x0, 0xc00109d928?}, 0x0, {0x8310ceb, 0x4})
	/Users/runner/work/tinygo/tinygo/interp/interpreter.go:544 +0x84b4
github.com/tinygo-org/tinygo/interp.Run({0xc00109ddf0?}, 0x3f?)
	/Users/runner/work/tinygo/tinygo/interp/interp.go:115 +0x6cc
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x600000010000?}, 0xc00016c120)
	/Users/runner/work/tinygo/tinygo/builder/build.go:1058 +0x33
github.com/tinygo-org/tinygo/builder.Build.func5(0xc003309c80)
	/Users/runner/work/tinygo/tinygo/builder/build.go:561 +0x5ed
github.com/tinygo-org/tinygo/builder.runJob(0xc003309ce0, 0x0?)
	/Users/runner/work/tinygo/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/runner/work/tinygo/tinygo/builder/jobs.go:123 +0x5db
```

```
$ echo Hello world > text.txt
```

```
package main

import _ "embed"

//go:embed text.txt
var rawWords string

var byteWords = []byte(rawWords)

func main() {
}
```

```
tinygo version 0.24.0 darwin/amd64 (using go version go1.18 and LLVM version 14.0.0)
Darwin 21.3.0 Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_ARM64_T6000 arm64
```

dgryski, 2022-08-02:
Looks like an issue with `interp`.

koodimetsa, 2022-08-25:
Same other way around.

codefromthecrypt, 2022-09-07:
So, `srcObject.buffer` is still nill when the coercion happens. I looked a little closer. if you cast inside `main` it works. There may be something lazy that needs to become eager or special case for embed expanded variables. Not sure.

fyi, the `-dumpssa` flag helps a bit as at least you can see the length (12) is correct.
```bash
$ tinygo build -dumpssa main.go
--snip--
call: embed.init
    ret
call: main.init
    call: runtime.stringToBytes(<273>, 12, 0)
        alloca: [24] -> <275>
        alloca: [16] -> <276>
        insertvalue: [<[…16]> <273> 0] -> <[…16]>
        insertvalue: [<[…16]> 12 8] -> <[…16]>
        gep: [<276> 8 0 16 0 -1] -> <276>
        store: 0 <276>
        gep: [<276> 8 0 16 8 -2] -> <276+8>
        store: 0 <276+8>
        store: <273> <276>
        store: 12 <276+8>
        gep: [<275> 8 0 24 0 -1] -> <275>
        store: 0 <275>
        gep: [<275> 8 0 24 8 -2] -> <275+8>
        store: 0 <275+8>
        gep: [<275> 8 0 24 16 -3] -> <275+16>
        store: 0 <275+16>
        br [0 9 1] -> 1
        load: <276+8> -> 12
        runtime.alloc: 12 -> <277>
        br [0 10 2] -> 2
        load: <276> -> <273>
        br [0 11 3] -> 3
        load: <276+8> -> 12
        call: runtime.memcpy(<277>, <273>, 12, 0)
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x42e5228]
```

codefromthecrypt, 2022-09-15:
verified this works now. thanks @aykevl!

closed 2022-09-29 by aykevl