tinygo-org/tinygo #3028: uIP == TCP/IP in cooperative multitasking

vanrein, 2022-07-27:
As suggested on MCH2022, you may want to look into the TCP/IP for 8/16 bit microcontrollers, really small and pretty feature complete.  There is now an IPv6 version too.

https://en.wikipedia.org/wiki/UIP_(micro_IP)

Oh, and you were surprised about cooperative multitasking in C, but it's done from time to time.  Putty is the best-known example I suppose.  It usually comes as very, very ugly header include files.

I really enjoyed seeing Go for microcontrollers, thank you for the idea and effort!

soypat, 2022-08-05:
Worked on this for a while rolling my own a while back for the ENC28J60 IC. Was working okish, managed to get a basic http webpage to control a LED on the arduino uno. Still needed quite a bit of work to be extensible. I'll be taking a look at UIP since I'm still interested in having this feature.

IC Driver: https://github.com/tinygo-org/drivers/pull/253
TCP/IP stack: https://github.com/soypat/ether-swtch

aykevl, 2022-08-31:
@vanrein thank you for the suggestion! I haven't worked much on the net package but will keep it in mind as an option.