tinygo-org/tinygo #3034: tinygo info results differ

sago35, 2022-07-29:
GOARM is 6 when run on ubuntu and 7 on windows.
Is this as intended?

```
# ubuntu 20.04
$ ./bin/tinygo version
tinygo version 0.25.0-dev-7b1e5f6 linux/amd64 (using go version go1.17.6 and LLVM version 14.0.0)

$ ./bin/tinygo info wioterminal
LLVM triple:       thumbv7em-unknown-unknown-eabi
GOOS:              linux
GOARCH:            arm
GOARM:             6
build tags:        cortexm baremetal linux arm atsamd51p19a atsamd51p19 atsamd51 sam wioterminal tinygo math_big_pure_go gc.conservative scheduler.tasks serial.usb
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     ...
```

```
# windows11
$ .\bin\tinygo.exe version
tinygo version 0.25.0-dev-7b1e5f6 windows/amd64 (using go version go1.18.3 and LLVM version 14.0.0)

$ .\bin\tinygo.exe info wioterminal
LLVM triple:       thumbv7em-unknown-unknown-eabi
GOOS:              linux
GOARCH:            arm
GOARM:             7
build tags:        cortexm baremetal linux arm atsamd51p19a atsamd51p19 atsamd51 sam wioterminal tinygo math_big_pure_go gc.conservative scheduler.tasks serial.usb
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     ...
```

sago35, 2022-07-29:
It seems to be branching here and it is 7 for windows and android.
https://github.com/tinygo-org/tinygo/blob/v0.25.0-beta1/goenv/goenv.go#L53-L66

aykevl, 2022-08-31:
Yes, `GOARM` is different but it has no effect on the binaries. It is only used when creating native binaries. For example:

```
$ GOOS=windows tinygo info ""
LLVM triple:       x86_64-unknown-windows-gnu
GOOS:              windows
GOARCH:            amd64
GOARM:             7
[...]

$ GOOS=linux tinygo info ""
LLVM triple:       x86_64-unknown-linux
GOOS:              linux
GOARCH:            amd64
GOARM:             6
[...]
```

So, I'm not sure what to do about it. We can for example simply remove the `GOARM` key?

sago35, 2023-02-21:
Closed because it has been resolved.

PR #3108 (2022-09-01): main: remove GOARM from `tinygo info`
I think it is more confusing than helpful because it is only relevant when compiling an actual linux/arm binary (and in that case, it is also included in the LLVM triple).

Fixes #3034
closed 2023-02-21 by sago35