tinygo-org/tinygo #3041: Adding support for Power Linux (ppc64le) platform.

Sunidhi-Gaonkar1, 2022-08-03:
Hi Team, I am creating this issue to request the community to provide build artifacts for Power Linux (ppc64le) platform.
Tinygo is being used by `aquasecurity/trivy` (https://github.com/aquasecurity/trivy) , to run unit and module-integration test cases. I am working on validating Trivy on Power(ppc64le) architecture.
I was able to build Tinygo from source following the instructions mentioned here: https://github.com/tinygo-org/tinygo/blob/release/BUILDING.md.
If there is anything I can do to help with build and testing, I can pick that up. Thank you.

dkegel-fastly, 2022-08-05:
I see Travis supports ppc64le... https://docs.travis-ci.com/user/multi-cpu-architectures/
and qemu could be used anywhere, e.g. https://github.com/marketplace/actions/run-on-architecture

aykevl, 2022-08-31:
@Sunidhi-Gaonkar1 just to be clear, are you interested in building a ppc64le `tinygo` binary, building ppc64le Go binaries using TinyGo, or both? Because these two tasks are mostly separated.

> I see Travis supports ppc64le... https://docs.travis-ci.com/user/multi-cpu-architectures/
> and qemu could be used anywhere, e.g. https://github.com/marketplace/actions/run-on-architecture

You are certainly correct, but I prefer to reduce the number of CIs that we use for building. We currently already use two (CircleCI and GitHub Actions). Also, we already produce a relatively large number of build artifacts and I haven't seen many people use PowerPC recently. So yeah, I'm not entirely sure about adding ppc64le build artifacts but we could do it (similar to how we've added arm and arm64 Linux builds).

Sunidhi-Gaonkar1, 2022-09-01:
@aykevl Thank you for your response,I am interested in building a ppc64le `tinygo` binary. As GitHub Actions and qemu is being used to add build artifacts for arm and arm64 Linux builds , It can also be used to add build artifacts for ppc64le.
If I can contribute for the same in any way, Please let me know.