tinygo-org/tinygo #3044: tinygo cgo compile error

halfrost, 2022-08-03:
hen I use tinygo to compile wasm plugin: 

```
$ tinygo build -o main.wasm -scheduler=none -target=wasi main.go
```
compile error. The error log is:

```
# os/user
../../../../opt/Cellar/go/1.17.2/libexec/src/os/user/cgo_lookup_unix.go:21:6: not implemented: build constraints in #cgo line
../../../../opt/Cellar/go/1.17.2/libexec/src/os/user/cgo_lookup_unix.go:24:10: fatal: 'pwd.h' file not found
../../../../opt/Cellar/go/1.17.2/libexec/src/os/user/getgrouplist_unix.go:15:10: fatal: 'grp.h' file not found
```

Once I disabled the cgo compile parameters, 

```
$ CGO_ENABLED=0 tinygo build -o main.wasm -scheduler=none -target=wasi main.go
```

The error log is:

```
# github.com/cespare/xxhash/v2
../../pkg/mod/github.com/cespare/xxhash/v2@v2.1.1/xxhash_unsafe.go:32:11: cannot use len(s) (value of type int) as uintptr value in assignment
../../pkg/mod/github.com/cespare/xxhash/v2@v2.1.1/xxhash_unsafe.go:33:11: cannot use len(s) (value of type int) as uintptr value in assignment
../../pkg/mod/github.com/cespare/xxhash/v2@v2.1.1/xxhash_unsafe.go:43:11: cannot use len(s) (value of type int) as uintptr value in assignment
../../pkg/mod/github.com/cespare/xxhash/v2@v2.1.1/xxhash_unsafe.go:44:11: cannot use len(s) (value of type int) as uintptr value in assignment
```

My dependencies include prometheus and etcd. Both of their dependencies include xxhash. Is there any good idea?

dgryski, 2022-08-03:
Try building with `-tags=appengine` to get a safe version of xxhash.

halfrost, 2022-08-03:
@dgryski I try this command:

```
CGO_ENABLED=0 tinygo build -o main.wasm -scheduler=none -tags=appengine -target=wasi main.go
```

But still error:

```
# github.com/golang/protobuf/proto
../../pkg/mod/github.com/golang/protobuf@v1.4.3/proto/registry.go:230:16: MapOf not declared by package reflect
```

dgryski, 2022-08-03:
Yup, that's a current limitation of the `reflect` package.   There are patches in progress to fix that, but otherwise no timeline for getting them merged.

halfrost, 2022-08-03:
OK, I'll keep an eye on your patch. You can add a tag to this issue with the patch you mentioned.

cross-hello, 2022-12-30:
@dgryski same problem here.
[image]
https://github.com/XTLS/Xray-core/issues/1473#issuecomment-1367775901

davecheney, 2026-10-26:
Sorry for the delay in replying. Some of the reported reflection limitations have since been implemented, but protobuf support remains incomplete.

A small protobuf marshal/unmarshal example still fails with TinyGo 0.44-dev at runtime with `unimplemented: (reflect.Type).MethodByName()`. The sample and commands are posted at https://github.com/tinygo-org/tinygo/issues/1879#issuecomment-6408872332.

Leaving this open. If your application still fails with TinyGo 0.43 or later, please share a minimal reproducer and your dependency versions so we can identify the remaining blockers.