tinygo-org/tinygo #3057: tinygo-test fails with undefined symbols

QuLogic, 2022-08-06:
With Go 1.18, I am getting the following error when running `make tinygo-test`:
```
ld.lld: error: undefined symbol: getpagesize
>>> referenced by os_linux.go:122 (/builddir/build/BUILDROOT/tinygo-0.25.0-1.fc36.x86_64/usr/lib64/tinygo/src/runtime/os_linux.go:122)
>>>               /tmp/tinygo863360999/main.o:(os_test.TestGetpagesize)
error: failed to link /tmp/tinygo863360999/main: exit status 1
```
With Go 1.19, I additionally get the following error on any test that uses crypto:
```
ld.lld: error: undefined symbol: crypto/internal/boring/sig.StandardCrypto
>>> referenced by notboring.go:25 (/usr/lib/golang/src/crypto/internal/boring/notboring.go:25)
>>>               /builddir/build/BUILD/tinygo-0.25.0/tinygo.OFH7PV/tinygo/thinlto/llvmcache-51407FBE491F17C7777BEE75E99D1816E029D4F3:(crypto/internal/boring.Unreachable)
error: failed to link /tmp/tinygo3462869915/main: exit status 1
```

This started with tinygo 0.24.0. It seemed to be related to CGo, so I tried reverting #2774, but that did not help. I also tried force disabling ThinLTO, but that did not help either. So I'm running low on ideas. I am running with external LLVM (14.0.5 against Go 1.19, and 14.0.0 against Go 1.18)

QuLogic, 2022-08-06:
Though that being said, 0.23.0 never compiled with Go 1.19, so I can't say if it wouldn't have brought up the same errors. Similarly, the test for `getpagesize` was only added in 0.24.0, so I can't say it also wouldn't have errored in earlier versions.

So I don't know if this actually regressed from 0.23.0 to 0.24.0 or there are just more tests that show it was already broken.

QuLogic, 2022-08-06:
So I backported #2899 to 0.23.0 and it still failed, so this seems to have always been broken and there just wasn't anything that triggered it?

dkegel-fastly, 2022-08-18:
Could it be that getpagesize() is no longer available everywhere?  
Evidently it is deprecated, see e.g. https://refspecs.linuxbase.org/LSB_4.1.0/LSB-Core-generic/LSB-Core-generic/baselib-getpagesize.html

aykevl, 2022-08-31:
@QuLogic I can't seem to reproduce the `getpagesize` issue on the latest dev branch:

```
$ tinygo test os
ok      os      0.080s

$ tinygo version
tinygo version 0.26.0-dev linux/amd64 (using go version go1.19 and LLVM version 14.0.6)
```

Was it the `os` package that was failing for you?

Also, if you are using the Fedora package, are you sure the musl library is built correctly? It should have a `getpagesize` symbol.

> Could it be that getpagesize() is no longer available everywhere?

We use musl, so if it's missing it's because something is wrong with it.

QuLogic, 2022-10-08:
I just tried with 0.26.0 and it is still failing. I don't use any pre-compiled musl as tinygo needs it for all the unique architectures it targets that may not be the same as Fedora's. The only difference from the builds here should be that I pull musl from its 1.2.0 release tarball instead of the git commit. Everything else should be the same, but I'll try looking at the musl parts a bit closer.

aykevl, 2022-10-09:
I found the issue, working on a fix in https://github.com/tinygo-org/tinygo/pull/3223.

QuLogic, 2022-10-09:
Yes, I came to the same conclusion. I don't use the `Makefile` directly, but do copy what is done in the `build/release` target (I do have some plans to make it reusable for me, but haven't gotten around to it.) And when I run tests, it's against the to-be-installed `TINYGOROOT`, not the source tree. Since it's found with a glob, tinygo doesn't care that the `legacy` directory is missing, but as it contains `getpagesize`, then it fails during linking.

aykevl, 2022-10-09:
> Since it's found with a glob, tinygo doesn't care that the `legacy` directory is missing, [...]

Ah of course, that's where I need to insert a check! I have updated #3223 accordingly, thanks for the hint.

achille-roussel, 2023-01-07:
Hello,

I believe I am running into another occurrence of this issue, here are the steps I used:

- clone a fork of tingyo form my Github https://github.com/achille-roussel/tinygo
- install dependencies as instructed in the "getting started"
- `make llvm-source llvm-build`
- `make`
- `./build/tinygo test os`

I am getting different errors on Linux and Darwin, tho they seem to be somewhat similar linking issues.

* **Linux**
```
$ ./build/tinygo version
tinygo version 0.27.0-dev-0298b207 linux/amd64 (using go version go1.18.1 and LLVM version 15.0.0)
$ ./build/tinygo test os
tinygo:ld.lld: error: undefined symbol: getpagesize
>>> referenced by os_linux.go:121 (/home/achille_roussel/go/src/github.com/achille-roussel/tinygo/src/runtime/os_linux.go:121)
>>>               /home/achille_roussel/.cache/tinygo/thinlto/llvmcache-73CCF93F09B5C922F351809D4AF8603DB5962F3A:(os_test.TestGetpagesize)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo3497474776/main: exit status 1
```

* **Darwin**
```
$ ./build/tinygo version
tinygo version 0.27.0-dev-9fd0567f darwin/arm64 (using go version go1.19 and LLVM version 15.0.0)
$ ./build/tinygo test os
tinygo:ld.lld: error: undefined symbol: _syscall.closedir
>>> referenced by dir_darwin.go:23 (/Users/achilleroussel/dev/src/github.com/achille-roussel/tinygo/src/os/dir_darwin.go:23)
>>>               /Users/achilleroussel/Library/Caches/tinygo/thinlto/llvmcache-3628447F07707191BB435FF0277F81E948339990:(symbol _(*os.File).Close+0x34)
failed to run tool: ld.lld
error: failed to link /var/folders/dd/pdhyqqvs0pn9llv76d27vt240000gn/T/tinygo2678297289/main: exit status 1
```

I am working off of the latest version of the `dev` branch, which includes the fix mentioned above from #3223

achille-roussel, 2023-01-07:
The issue on Darwin ended being caused by an issue in the code change I was making in #3370, which I addressed with https://github.com/tinygo-org/tinygo/pull/3370/commits/0de35def867b19335a131a2f28733b6b3869435f Because it was a linker issue I mistakenly made the assumption that it might be related to the linking program on Linux.

I am still investigating the linker error on Linux, will update if I find a workaround.

achille-roussel, 2023-01-09:
🤔 I'm currently not able to reproduce the issue. It's unclear to me what addressed the issue, the main thing I changed is upgrading from Go 1.18 to 1.19, tho I don't know how that may be related.

aykevl, 2023-01-10:
Let's close it then as the original issue is addressed. @achille-roussel if you find a bug like this again, please open a new issue.

PR #3223 (2022-10-16): ci: include 'getpagesize' function in release
I wonder why nobody else noticed this bug? In any case, this patch fixes it.

Fixes #3057.

(This is a draft, the fix is a trivial one-line fix but I want to make sure the other changes indeed test this change).
closed 2023-01-10 by aykevl