tinygo-org/tinygo #3062: Basic Block in generics test does not have terminator

QuLogic, 2022-08-08:
This seems to be a flaky error. I had to repeat builds 3 times until it passed, and the error was slightly different each time. The failure is something like:
```
Basic Block in function 'github.com/tinygo-org/tinygo/testdata/generics/value.New[int]' does not have terminator!
label %entry
    main.go:1195: error: verification error after compiling package github.com/tinygo-org/tinygo/testdata/generics/testa
```
The reason why I say it's flaky is that [this first build errors on `EmulatedCortexM3`](https://kojipkgs.fedoraproject.org//work/tasks/6304/90586304/build.log), but [the second build errors on `Host`](https://kojipkgs.fedoraproject.org//work/tasks/8375/90588375/build.log), while [the third build is fine](https://kojipkgs.fedoraproject.org//work/tasks/1579/90591579/build.log).

aykevl, 2022-08-13:
Hmm, this suggests some sort of memory error or race condition in the compiler. Can you try rebuilding LLVM with assertions enabled? It might surface the actual bug.

deadprogram, 2023-05-15:
Is this still an issue @QuLogic ?

QuLogic, 2023-08-28:
Unfortunately yes, I just hit this while testing out 0.29.0:
```
Basic Block in function '(*github.com/tinygo-org/tinygo/testdata/generics/value.mapper[int, int]).Get[int int]' does not have terminator!
label %entry
=== NAME  TestBuild/WebAssembly/generics.go
    main.go:1320: error: verification error after compiling package github.com/tinygo-org/tinygo/testdata/generics/testa
```

aykevl, 2024-04-17:
I'm pretty sure this specific issue (crashes when compiling generic code) has been fixed in https://github.com/tinygo-org/tinygo/issues/3895.
We're now seeing similar issues in https://github.com/tinygo-org/tinygo/issues/4206 but that appears to be a different problem because it also affects non-generic code.

closed 2024-04-17 by aykevl