tinygo-org/tinygo #3067: can I build tinygo project as a static link file and link it to another C project?

jifanchn, 2022-08-10:
As described above.

I'm wondering if we can just create a very tiny bootstrap C project and link it to a tinyGO program?

And in that case, tinygo output will be "xxx.h" and "xxx.so" and finally link to a basic C project which prepare working environments for tinygo core.

So that maybe we can use most currently useful C platform code and handle data with tinygo.

Nor don't we need to write platform hardware related code in tinygo and tinygo itself can run as a VM(runtime).

aykevl, 2022-08-31:
You can compile to an object file by giving it an `.o` extension. But it will likely lack some symbols you need.
Alternatively, you can use Go and include external C files using CGo.

Can you give a bit more detail what you're trying to accomplish? What hardware are you compiling for, and with what environment?

aykevl, 2023-05-03:
This is not really a supported configuration. TinyGo is its own RTOS, it does not currently support linking with an existing RTOS.
This might get added in the future, but we currently have no plans for this.

davecheney, 2026-10-24:
Thanks for the question. Linking TinyGo into an existing RTOS is not currently supported. Zephyr integration is tracked in #379.

closed 2026-10-24 by davecheney