tinygo-org/tinygo #3071: Missing pprof.WriteHeapProfile

ntrrg, 2022-08-14:
I was going to create a pull request just adding this in `src/runtime/pprof/pprof.go`, but didn't know if it would be problematic.

```go
func WriteHeapProfile(w io.Writer) error {
	return nil
}
```

With that, this code would work fine in Go and TinyGo.

```go
package main

import (
	"os"
	"runtime/pprof"
)

func main() {
	if cpup := os.Getenv("CPUPROFILE"); cpup != "" {
		f, err := os.Create(cpup)
		if err != nil {
			fmt.Fprintf(os.Stderr, "cannot create CPU profile file: %v\n", err)
			os.Exit(1)
		}

		defer f.Close()

		if err := pprof.StartCPUProfile(f); err != nil {
			fmt.Fprintf(os.Stderr, "cannot profile CPU usage: %v\n", err)
			os.Exit(1)
		}

		defer pprof.StopCPUProfile()
	}

	// Extra code here.

	if memp := os.Getenv("MEMPROFILE"); memp != "" {
		f, err := os.Create(memp)
		if err != nil {
			fmt.Fprintf(os.Stderr, "cannot create memory profile file: %v\n", err)
			os.Exit(1)
		}

		defer f.Close()

		if err := pprof.WriteHeapProfile(f); err != nil {
			fmt.Fprintf(os.Stderr, "cannot profile memory usage: %v\n", err)
			os.Exit(1)
		}
	}
}
```

dgryski, 2022-08-14:
Do we want to return `nil` or an error instead?

ntrrg, 2022-08-15:
I guess `nil` since `pprof.StartCPUProfile` do the same

dgryski, 2022-08-16:
Yes. Go ahead and make a PR with this change.

PR #3076 (2022-08-16): runtime/pprof: add WriteHeapProfile
Fixes: #3071
commit d0808c93 (2022-08-16): runtime/pprof: add WriteHeapProfile

Fixes: #3071
closed 2022-09-29 by aykevl