tinygo-org/tinygo #3077: package runtime/volatile: missing function body

goelvatsal, 2022-08-16:
Plugin version: TinyGo 0.25.0
IDE name and version: IntelliJ GoLand 2022.2.1
OS name: MacOS Monterey

GOPATH: /Users/goel01/go/
GOROOT (if needed): /usr/local/Cellar/go/1.18.5/libexec/
Project location: /Users/goel01/go/src/github.com/PacktPublishing/Creative-DIY-Microcontroller-Projects-with-TinyGo-and-WebAssembly

I'm trying to run a program called [water-level-sensor-example](https://github.com/PacktPublishing/Creative-DIY-Microcontroller-Projects-with-TinyGo-and-WebAssembly/tree/master/Chapter04/water-level-sensor-example), with [water-level-sensor](https://github.com/PacktPublishing/Creative-DIY-Microcontroller-Projects-with-TinyGo-and-WebAssembly/tree/master/Chapter04/water-level-sensor) being a library for the former. My IDE (GoLand) asks me to download a package called runtime/volatile. After downloading it, I go to run the program again, and it outputs 8 errors stating that there is a "missing function body". Thanks in advance. I can provide more information in case you need it.

[image]

deadprogram, 2022-08-16:
This looks like an issue with the GoLand plugin, which I am not really very familiar with. Perhaps it is not configured properly? I'd suggest you use the resources listed here https://plugins.jetbrains.com/plugin/16915-tinygo to contact the maintainers of the plugin.

goelvatsal, 2022-09-04:
> This looks like an issue with the GoLand plugin, which I am not really very familiar with. Perhaps it is not configured properly? I'd suggest you use the resources listed here https://plugins.jetbrains.com/plugin/16915-tinygo to contact the maintainers of the plugin.

Sorry for the very late response, I can't seem to find the managers of the TinyGo plugin on the website, it just says that the vendor is JetBrains.

dgryski, 2022-09-05:
/cc @dlsniper

nickfromXXII, 2022-09-05:
@goelvatsal Yes, it seems to be a plugin issue. It would be nice if you could provide more information about your problem by submitting a ticket on the JetBrains tracker at https://youtrack.jetbrains.com/issues/GO. Thanks!

nickfromXXII, 2022-09-05:
@goelvatsal Sorry for inconvenience, it seems to be a plugin issue. It would be nice if you could provide concrete steps for reproducing the problem by submitting a ticket on the JetBrains tracker at https://youtrack.jetbrains.com/issues/GO. Thank you!

deadprogram, 2022-09-10:
Closing as specific to JetBrains plugin. Please open a ticket there as requested. Thank you!

closed 2022-09-10 by deadprogram