tinygo-org/tinygo #3078: Execute code from RAM, on Raspberry pi Pico

0xcafed00d, 2022-08-17:
Is there currently anyway to mark a TinyGo function as executing from RAM? Something like the //go:inline directive?
If not is it possible then is there any way to mark a cgo function as executing from RAM?

I've tried to mark reproduce the `__no_inline_not_in_flash_func` macro used in the pico sdk, but the code doesnt seem to get put the the correct location in the elf file produced...

```c++
#define __not_in_flash(group) __attribute__((section(".time_critical." group)))
#define __not_in_flash_func(func_name) __not_in_flash(__STRING(func_name)) func_name
#define __no_inline_not_in_flash_func(func_name) __attribute__((noinline)) __not_in_flash_func(func_name)
```

aykevl, 2022-09-08:
Correct, we don't support anything like this at the moment. Perhaps we should.
Here is a similar issue: https://github.com/tinygo-org/tinygo/issues/2599, and I'm sure I've seen this come up before.

aykevl, 2022-09-08:
Also related: https://github.com/tinygo-org/tinygo/pull/1759

closed 2023-06-14 by deadprogram