tinygo-org/tinygo #3082: Modified tm1637 driver

placessweden, 2022-08-19:
I just couldn't get the tm1637 driver to work on any board but Arduino Uno (AVR).
After much headscratching I came up with a modified version, where the dio and
clk pins are not bidirectional but only configured as output, then using High() and Low().
I also modified the way bits are masked i method writeByte().
I've tested the code on RPI Pico, Trinket-M0, Sipeed Maix Bit and Arduino Uno.
Code and diff:
https://github.com/placessweden/Modified-tm1637-for-TinyGo

deadprogram, 2022-08-19:
Wouldn't this be better off in the Drivers repo? https://github.com/tinygo-org/drivers/issues

placessweden, 2022-08-19:
Probably, but I'm a rookie and don't know how to commit - and have only had a GitHub account for 23 hours now :)

deadprogram, 2022-08-19:
IN any case, please open an issue in that repo. Thanks! Now closing.

closed 2022-08-19 by deadprogram