tinygo-org/tinygo #3083: compiler panics on unsafe.Offsetof of chan field

ALTree, 2022-08-19:
```
$ tinygo version
tinygo version 0.25.0 linux/amd64 (using go version go1.18.3 and LLVM version 14.0.0)
```

```
package main

import "unsafe"

func main() {
	var s struct{ f chan int }
	println(unsafe.Offsetof(s.f))
}
```

```
$ tinygo build crash.go
panic: unknown type: chan int [recovered]
	panic: unknown type: chan int

goroutine 1 [running]:
go/types.(*Checker).handleBailout(0xc0001ea000, 0xc000548a98)
	/__t/go/1.18.1/x64/src/go/types/check.go:303 +0x8b
panic({0x51d9da0, 0xc000324b70})
	/__t/go/1.18.1/x64/src/runtime/panic.go:838 +0x207
github.com/tinygo-org/tinygo/compiler.(*stdSizes).Sizeof(0xc0000285b8, {0x52c54f0?, 0xc000878138?})
	/__w/tinygo/tinygo/compiler/sizes.go:161 +0x387
github.com/tinygo-org/tinygo/compiler.(*stdSizes).Alignof(0xc0000285b8, {0x52c54f0, 0xc000878138})
	/__w/tinygo/tinygo/compiler/sizes.go:53 +0x17e
github.com/tinygo-org/tinygo/compiler.(*stdSizes).Offsetsof(0x52c54f0?, {0xc0009d80e0, 0x1, 0xc000878138?})
	/__w/tinygo/tinygo/compiler/sizes.go:72 +0xdd
go/types.(*Config).offsetsof(0x52c5608?, 0xc00031bc50)
	/__t/go/1.18.1/x64/src/go/types/sizes.go:229 +0x5c
go/types.(*Config).offsetof(0x52c5608?, {0x52c5608?, 0xc00031bc50?}, {0xc0002832f0, 0x1, 0x0?})
	/__t/go/1.18.1/x64/src/go/types/sizes.go:253 +0xa5
go/types.(*Checker).builtin(0xc0001ea000, 0xc0007ee940, 0xc00022f280, 0x11)
	/__t/go/1.18.1/x64/src/go/types/builtins.go:707 +0x4625
...
```

dgryski, 2022-08-19:
Thanks for the nice small reproduction case

aykevl, 2022-08-31:
Yes, that's a very nice reproducer, thank you!
I have fixed the issue here: https://github.com/tinygo-org/tinygo/pull/3112

closed 2022-09-29 by aykevl