tinygo-org/tinygo #3087: wasm: implement readdir

codefromthecrypt, 2022-08-22:
Until readdir is implemented in wasm, we can't re-use go's utilities for testing for bugs in impl.
```
2022/01/01 00:00:00 TestFS err: TestFS found errors:
        .: ReadDir(-1): readdir unimplemented : errno 54
```

Here's what would be nice to be able to both compile and run (right now, it compiles but can't run).
```go
package main

import (
	"fmt"
	"log"
	"os"
	"syscall"
	"testing/fstest"
)

func main() {
	if err := fstest.TestFS(os.DirFS("sub"), "test.txt"); err != nil {
		log.Panicln("TestFS err:", err)
	}
	fmt.Println("TestFS ok")
}
```

deadprogram, 2023-05-15:
https://github.com/tinygo-org/tinygo/pull/3696 addresses this issue. I have tagged it to be closed on next release.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!