tinygo-org/tinygo #3093: reflect.SliceHeader uses uintptr for Len and Cap instead of int

dgryski, 2022-08-24:
https://github.com/golang/go/blob/master/src/reflect/value.go#L2695

vs

https://github.com/tinygo-org/tinygo/blob/release/src/reflect/value.go#L773

This causes a build failure for https://github.com/goccy/go-json

```
internal/encoder/string.go:20:9: cannot use len(s) / 8 (value of type int) as uintptr value in struct literal
```

aykevl, 2022-08-24:
Please take a look here: https://github.com/tinygo-org/tinygo/issues/1284

I believe this is a duplicate, and there is nothing to fix.

dgryski, 2022-08-24:
Closing.  I'll try to come up with a patch for `goccy/go-json` to make it work on both implementations.

closed 2022-08-24 by dgryski

dgryski, 2022-08-24:
```
 func stringToUint64Slice(s string) []uint64 {
-       return *(*[]uint64)(unsafe.Pointer(&reflect.SliceHeader{
-               Data: ((*reflect.StringHeader)(unsafe.Pointer(&s))).Data,
-               Len:  len(s) / 8,
-               Cap:  len(s) / 8,
-       }))
+       hdr := *(*reflect.StringHeader)(unsafe.Pointer(&s))
+       return unsafe.Slice((*uint64)(unsafe.Pointer(hdr.Data)), len(s)/8)
 }
```

twmb, 2022-08-31:
The one change I would make to the prior comment is
```diff
- hdr := *(*reflect.StringHeader)(unsafe.Pointer(&s))
+ hdr := (*reflect.StringHeader)(unsafe.Pointer(&s))
```

aykevl, 2022-08-31:
Just so you know: casting to `[]uint64` can result in unaligned accesses. This can lead to problems on most architectures ([including amd64](https://pzemtsov.github.io/2016/11/06/bug-story-alignment-on-x86.html)!).
Also, writing to a `[]uint64` that used to be a string is very much undefined behavior, but I guess you already knew that ;)