tinygo-org/tinygo #3096: tinygo build panic appeared

FGadvancer, 2022-08-26:
hi, guys
     When I try to compile the wasm module with tinygo, the following panic message appears(environment：mac Mojave10.14.6,go version 1.17.8 darwin/amd64,tinygo,version 0.25.0):

panic: runtime error: invalid memory address or nil pointer dereference [recovered]
	panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x10 pc=0x428b234]

goroutine 1 [running]:
golang.org/x/tools/go/ast/astutil.Apply.func1()
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:48 +0x89
panic({0x82b8ec0, 0x9a303b0})
	/Users/runner/hostedtoolcache/go/1.18.1/x64/src/runtime/panic.go:838 +0x207
github.com/tinygo-org/tinygo/cgo.(*cgoPackage).isEquivalentAST(0x82a3000?, {0x8392290?, 0x0?}, {0x8392290?, 0x0})
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:944 +0x434
github.com/tinygo-org/tinygo/cgo.(*cgoPackage).isEquivalentAST(0x8?, {0x8392308?, 0xc004d90060?}, {0x8392308?, 0xc004b92b10})
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:991 +0x2db
github.com/tinygo-org/tinygo/cgo.(*cgoFile).getASTDeclNode(0xc000362a98, {0xc004a51790, 0x4}, {0x8, 0x0, {0x13903a80, 0x1, 0xc800f80}}, 0x7?)
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:1140 +0x9c5
github.com/tinygo-org/tinygo/cgo.(*cgoFile).getASTDeclName(0x0?, {0xc004a51790, 0x4}, {0x8, 0x0, {0x13903a80, 0x1, 0xc800f80}}, 0x1)
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:1119 +0xc5
github.com/tinygo-org/tinygo/cgo.(*cgoFile).walker(0x36?, 0x16?, 0xc0038dd430?)
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:1228 +0x157
github.com/tinygo-org/tinygo/cgo.Process.func3.1(0x4?)
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:315 +0x27
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x8392240?, 0xc004a6d610?}, {0x83101ff?, 0x0?}, 0x8?, {0x8392088?, 0xc004a867c0?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:199 +0x202
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x8392038?, 0xc004a88060?}, {0x83106f7?, 0x0?}, 0xc004a88060?, {0x8392240?, 0xc004a6d610?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:327 +0x39b
golang.org/x/tools/go/ast/astutil.(*application).applyList(0xc004b73020, {0x8392038?, 0xc004a88060}, {0x83106f7, 0x4})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:484 +0xae
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x83923f8?, 0xc004a86800?}, {0x8310697?, 0x0?}, 0x8?, {0x8392038?, 0xc004a88060?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:353 +0x310
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x8392038?, 0xc004a88090?}, {0x83106f7?, 0x0?}, 0xc004a88090?, {0x83923f8?, 0xc004a86800?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:358 +0x4bf
golang.org/x/tools/go/ast/astutil.(*application).applyList(0xc004b73020, {0x8392038?, 0xc004a88090}, {0x83106f7, 0x4})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:484 +0xae
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x83925b0?, 0xc004a70b90?}, {0x8310697?, 0x0?}, 0x8?, {0x8392038?, 0xc004a88090?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:353 +0x310
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x8392038?, 0xc004a880c0?}, {0x83106f7?, 0x0?}, 0xc004a880c0?, {0x83925b0?, 0xc004a70b90?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:392 +0xf7b
golang.org/x/tools/go/ast/astutil.(*application).applyList(0xc004b73020, {0x8392038?, 0xc004a880c0}, {0x83106f7, 0x4})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:484 +0xae
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x8392308?, 0xc004a880f0?}, {0x8310697?, 0x0?}, 0x18?, {0x8392038?, 0xc004a880c0?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:353 +0x310
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x83922b8?, 0xc004a84500?}, {0x8310ad2?, 0x0?}, 0xc004a84500?, {0x8392308?, 0xc004a880f0?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:429 +0x1a29
golang.org/x/tools/go/ast/astutil.(*application).applyList(0xc004b73020, {0x83922b8?, 0xc004a84500}, {0x8310ad2, 0x5})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:484 +0xae
golang.org/x/tools/go/ast/astutil.(*application).apply(0xc004b73020, {0x8392ab0?, 0xc004e48be0?}, {0x831070f?, 0x1435a648?}, 0x18?, {0x83922b8?, 0xc004a84500?})
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:435 +0xc8d
golang.org/x/tools/go/ast/astutil.Apply({0x83922b8?, 0xc004a84500}, 0xc004c6adb0, 0x0)
	/Users/runner/go/pkg/mod/golang.org/x/tools@v0.1.11/go/ast/astutil/rewrite.go:53 +0x166
github.com/tinygo-org/tinygo/cgo.Process.func3(0xc004d90420)
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:314 +0xbe
github.com/tinygo-org/tinygo/cgo.(*cgoFile).readNames(0xc000362a98, {0xc0001306e0, 0x146}, {0xc001cf1340?, 0x18, 0x2c}, {0xc0049c143d, 0xb}, 0xc0038de820)
	/Users/runner/work/tinygo/tinygo/cgo/libclang.go:187 +0x562
github.com/tinygo-org/tinygo/cgo.Process({0xc004a84900, 0xf, 0x10?}, {0xc000024104, 0x32}, 0xc0018662c0, {0xc004ae3400, 0x9, 0xc000024104?}, {0xc00012c270, ...})
	/Users/runner/work/tinygo/tinygo/cgo/cgo.go:307 +0xb85
github.com/tinygo-org/tinygo/loader.(*Package).parseFiles(0xc00177fa40)
	/Users/runner/work/tinygo/tinygo/loader/loader.go:448 +0x2bf
github.com/tinygo-org/tinygo/loader.(*Package).Parse(0xc00177fa40)
	/Users/runner/work/tinygo/tinygo/loader/loader.go:363 +0x7f
github.com/tinygo-org/tinygo/loader.(*Program).Parse(0xc002733790)
	/Users/runner/work/tinygo/tinygo/loader/loader.go:312 +0x65
github.com/tinygo-org/tinygo/builder.Build({0x7ffeefbffa07, 0x9}, {0x7ffeefbff9f0, 0x9}, 0xc00020e2d0, 0xc0038df7b0)
	/Users/runner/work/tinygo/tinygo/builder/build.go:202 +0x11e5
main.Build({0x7ffeefbffa07, 0x9}, {0x7ffeefbff9f0, 0x9}, 0xc0001782c0)
	/Users/runner/work/tinygo/tinygo/main.go:156 +0x18c
main.main()
	/Users/runner/work/tinygo/tinygo/main.go:1441 +0x3b95

someone can answer my question? Thanks for all.

codefromthecrypt, 2022-09-07:
https://github.com/tinygo-org/tinygo/pull/3136 should fix this

deadprogram, 2022-09-30:
Released as part of `v0.26.0`

PR #3136 (2022-09-26): cgo: fixes panic when FuncType.Results is nil
FuncType.Results can be nil. This fixes the comparison and backfills relevant tests.

Fixes #3096
closed 2022-09-30 by deadprogram