tinygo-org/tinygo #3104: panic: unimplemented: (reflect.Value).MapRange()

kishorekonjeti, 2022-08-30:
I haven't used any relection in my code below. But still getting the issue. But the same working fine with official go compiler.
not sure even for small lines of code also its not working.
Can some one look on to this.?

- OS: ubuntu
- tinygo : version  0.25
- go version : 1.19

`
package main

import "fmt"

func main() {

	// Creating and initializing a map
	m_a_p := map[int]string{
		90: "Dog",
		}

	fmt.Println("Original map: ", m_a_p)

	// Checking the key is available
	// or not in the m_a_p map
	pet_name, ok := m_a_p[90]
	fmt.Println("\nKey present or not:", ok)
	fmt.Println("Value:", pet_name)
	 
}
`
$ tinygo build map_chk.go 
./map_chk
getting error message as 
**panic: unimplemented: (reflect.Value).MapRange()
Aborted (core dumped)**

aykevl, 2022-09-08:
You are actually using reflection, here:

```go
	fmt.Println("Original map: ", m_a_p)
```

The `fmt` package uses reflection to read the values it is given. But because `MapRange` isn't implemented, it crashes. Removing this line will likely get your program to work.

dgryski, 2023-02-24:
My recent reflect work #3470 adds this call.

dgryski, 2023-03-23:
This now works:
```
~/go/src/github.com/dgryski/bug/m $ go run main.go
Original map:  map[90:Dog]

Key present or not: true
Value: Dog
~/go/src/github.com/dgryski/bug/m $ tinygo run main.go
Original map:  map[90:Dog]

Key present or not: true
Value: Dog
```

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

closed 2023-06-14 by deadprogram