tinygo-org/tinygo #3106: Flash pybadge or ArduinoNano33 freezes my laptop and restarts it

doniacld, 2022-08-31:
## Description of the problem

When I plug my microntroller through USB to my computer and I flash it, sometimes I have the following behaviour:
1. It freezes my computer
2. It restarts it 
3. It proposes me to report the error below

```
panic(cpu 0 caller 0xfffffe001cd379e4): Kernel data abort. at pc 0xfffffe001ddc599c, lr 0x23907e001ddc59d8 (saved state: 0xfffffe60ba1bb3e0)
	  x0:  0x0000000000000000 x1:  0xfffffe001feaa768  x2:  0xfffffe150e3a4034  x3:  0xfffffe150e3a6e20
	  x4:  0xfffffe150e3a6e00 x5:  0x0000000000000000  x6:  0x96cd7e001cc51c2c  x7:  0xfffffe001f30a160
	  x8:  0x0000000000000000 x9:  0xfffffe150f173f90  x10: 0x00000000000032c0  x11: 0x0000000000004001
	  x12: 0x0001af1129900000 x13: 0x0000000000000154  x14: 0xfffffe150e879808  x15: 0x00000000004b0155
	  x16: 0xfffffe001f309740 x17: 0x1ba6fe001f309740  x18: 0x0000000000000000  x19: 0x00000000e00002ed
	  x20: 0xfffffe60ba1bb7f8 x21: 0xfffffe152fecfba0  x22: 0x0000000000000000  x23: 0x0000000000000001
	  x24: 0x37fdfe001dd5a8ac x25: 0x0000000000000000  x26: 0xfffffe60ba1bb8ae  x27: 0xcda1fe150f049d10
	  x28: 0xfffffe15349d2238 fp:  0xfffffe60ba1bb7b0  lr:  0x23907e001ddc59d8  sp:  0xfffffe60ba1bb730
	  pc:  0xfffffe001ddc599c cpsr: 0x60401208         esr: 0x96000006          far: 0x0000000000000000

Debugger message: panic
Memory ID: 0x1
OS release type: User
OS version: 21A559
Kernel version: Darwin Kernel Version 21.1.0: Wed Oct 13 17:33:01 PDT 2021; root:xnu-8019.41.5~1/RELEASE_ARM64_T6000
Fileset Kernelcache UUID: 3B2CA3833A09A383D66FB36667ED9CBF
Kernel UUID: 67BCB41B-BAA4-3634-8E51-B0210457E324
iBoot version: iBoot-7429.41.5
secure boot?: YES
Paniclog version: 13
KernelCache slide: 0x0000000014bcc000
KernelCache base:  0xfffffe001bbd0000
Kernel slide:      0x00000000153f4000
Kernel text base:  0xfffffe001c3f8000
Kernel text exec slide: 0x00000000154dc000
Kernel text exec base:  0xfffffe001c4e0000
mach_absolute_time: 0xc488e0653a6
Epoch Time:        sec       usec
  Boot    : 0x62f95428 0x00066e50
  Sleep   : 0x630f022e 0x0006aa42
  Wake    : 0x630f0522 0x0009c3fb
  Calendar: 0x630f1a7b 0x000bbe13

Zone info:
Foreign   : 0xfffffe00242e0000 - 0xfffffe00242f4000
Native    : 0xfffffe100044c000 - 0xfffffe300044c000
Readonly  : 0 - 0
Metadata  : 0xfffffe7ca8088000 - 0xfffffe7cb3ffc000
Bitmaps   : 0xfffffe7cb3ffc000 - 0xfffffe7cb9b7c000
CORE 0 PVH locks held: None
CORE 1 PVH locks held: None
CORE 2 PVH locks held: None
CORE 3 PVH locks held: None
CORE 4 PVH locks held: None
CORE 5 PVH locks held: None
CORE 6 PVH locks held: None
CORE 7 PVH locks held: None
CORE 8 PVH locks held: None
CORE 9 PVH locks held: None
CORE 0 is the one that panicked. Check the full backtrace for details.
CORE 1: PC=0xfffffe001c4f1b3c, LR=0xfffffe001c65534c, FP=0xfffffe60a6d63b70
CORE 2: PC=0xfffffe001c566c6c, LR=0xfffffe001c566c6c, FP=0xfffffe60b9813ef0
CORE 3: PC=0xfffffe001c4e0a44, LR=0xfffffe001c5c2f48, FP=0xfffffe60b941bd70
CORE 4: PC=0xfffffe001c566c70, LR=0xfffffe001c566c6c, FP=0xfffffe60ba373ef0
CORE 5: PC=0xfffffe001c566c70, LR=0xfffffe001c566c6c, FP=0xfffffe60b9d3bef0
CORE 6: PC=0xfffffe001c4f2fd4, LR=0xfffffe001c6763f4, FP=0xfffffe603943be50
CORE 7: PC=0xfffffe001c566c70, LR=0xfffffe001c566c6c, FP=0xfffffe60a730bef0
CORE 8: PC=0xfffffe001c664188, LR=0xfffffe001c664184, FP=0xfffffe60b9a63e80
CORE 9: PC=0xfffffe001c65d5d8, LR=0xfffffe001c548680, FP=0xfffffe60ba1cb570
Panicked task 0xfffffe152b744000: 79045 pages, 14 threads: pid 56772: tinygo
Panicked thread: 0xfffffe1530d9f440, backtrace: 0xfffffe60ba1baa90, tid: 6128746
[... 277 lines cut ...]
com.apple.driver.AppleAudioClockLibs	100.9.1
com.apple.driver.AppleM2ScalerCSCDriver	265.0.0
com.apple.iokit.IOSurface	302.9
com.apple.driver.IODARTFamily	1
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.security.AppleImage4	4.1.0
com.apple.kext.CoreTrust	1
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	12.0

** Stackshot Succeeded ** Bytes Traced 415751 (Uncompressed 1119024) **
```

## Solution

Using a USB hub prevents the problem from happening in a deterministic way.

deadprogram, 2022-08-31:
Probably related https://discussions.apple.com/thread/254012266

deadprogram, 2022-08-31:
Seems like https://forum.pjrc.com/threads/58547-Anyone-else-seeing-MacBook-Air-hard-crash-when-using-a-Teensy-4?p=299819&viewfull=1#post299819

deadprogram, 2022-10-20:
The fix for this was released in TinyGo `v0.26.0` so now closing. Please reopen if needed. Thank you!

rwwiv, 2022-12-30:
@deadprogram I'm still seeing this behavior on 0.26.0

```
OS: macOS 13.1 22C65 arm64 
Host: MacBookPro18,1 
Kernel: 22.2.0 
Packages: 268 (brew) 
Shell: zsh 5.8.1 
Terminal: vscode 
CPU: Apple M1 Pro 
GPU: Apple M1 Pro 
Memory: 3213MiB / 32768MiB 

> tinygo version
tinygo version 0.26.0 darwin/amd64 (using go version go1.19.4 and LLVM version 14.0.0)
```