tinygo-org/tinygo #3113: Tinygo support Arduino function millis() ?

yevsia, 2022-08-31:
Tinygo support Arduino function millis() ?

https://www.arduino.cc/reference/en/language/functions/time/millis/

How to implement it if possible...

sago35, 2022-09-02:
You can use time.Sleep.

yevsia, 2022-09-03:
I need to know the start time. time.Sleep this function like delay() in Arduino.

Arduino example code:
```
unsigned long myTime;

void setup() {
  Serial.begin(9600);
}
void loop() {
  Serial.print("Time: ");
  myTime = millis();

  Serial.println(myTime); // prints time since program started
  delay(1000);          // wait a second so as not to send massive amounts of data
}
```

deadprogram, 2022-09-05:
In Go you normally do this:

```go
start := time.Now()

for {
    if time.Since(start) > time.Second {
        // do your other thing
    }
    time.Sleep(time.Second)
}
```

deadprogram, 2022-09-10:
Closing as answered. Please reopen if needed.

closed 2022-09-10 by deadprogram