tinygo-org/tinygo #3114: Need help arduino PWM...

yevsia, 2022-08-31:
I'am use this example below for PWM but it's don't work, API changed. My experiments are not working correctly. 
https://create.arduino.cc/projecthub/alankrantas/tinygo-on-arduino-uno-an-introduction-6130f6

How to change this code to configure correctly.

`package main

import (
	"machine"
	"time"
)

func main() {

	machine.InitPWM()
	led := machine.PWM{machine.D9}
	led.Configure()

	duty := 1024

	for {

		led.Set(uint16(duty))

		time.Sleep(time.Millisecond * 25)
	}
}`

sago35, 2022-09-02:
@yevsia 
Does `src/examples/pwm` work?

yevsia, 2022-09-03:
In old API this code work fine:

```
package (
    "machine"
    "time"
)

func main() {

    machine.InitPWM()
    led := machine.PWM{machine.D9}
    led.Configure()

    duty, delta := 0, 1024

    for {

        led.Set(uint16(duty))
        duty += delta

        if duty < 0 || duty > 65535 {
            delta *= -1
            duty += delta
        }

        time.Sleep(time.Millisecond * 25)
     }
}
```

This code does not work in the new API.

I am using the new API but the code is not working properly :
```
package main

import (
	"machine"
	"time"
)

func main() {

	err := machine.Timer2.Configure(machine.PWMConfig{
		Period: 16384e3, // 16.384ms
	})
	if err != nil {
		println("failed to configure PWM")
		return
	}

	channelA, err := machine.Timer2.Channel(machine.PD3)
	if err != nil {
		println("failed to configure channel A")
		return
	}

	duty, delta := 0, 1024

	for {

		machine.Timer2.Set(channelA, uint32(duty))
		duty += delta

		if duty < 0 || duty > 65535 {
			delta *= -1
			duty += delta
		}

		time.Sleep(time.Millisecond * 25)
	}
}
```

yevsia, 2022-09-03:
I change PWM to machine.Timer2, pinB to machine.PD3 because I get errors:

PS E:\Development\Go\go> tinygo build -o src/TinyGo/PWM/main.hex -target=arduino src/TinyGo/PWM/main.go
# command-line-arguments
src\TinyGo\PWM\main.go:12:9: undeclared name: pwm
src\TinyGo\PWM\main.go:20:19: undeclared name: pwm
src\TinyGo\PWM\main.go:20:31: undeclared name: pinA
src\TinyGo\PWM\main.go:30:3: undeclared name: pwm

aykevl, 2022-09-15:
@yevsia can you share the code that you are compiling?

yevsia, 2022-09-16:
> package main

import (
	"machine"
	"time"
)

func main() {

	err := machine.Timer2.Configure(machine.PWMConfig{
		Period: 16384e3, // 16.384ms
	})
	if err != nil {
		println("failed to configure PWM")
		return
	}

	channelA, err := machine.Timer2.Channel(machine.PD3)
	if err != nil {
		println("failed to configure channel A")
		return
	}

	duty, delta := 0, 1024

	for {

		machine.Timer2.Set(channelA, uint32(duty))
		duty += delta

		if duty < 0 || duty > 65535 {
			delta *= -1
			duty += delta
		}

		time.Sleep(time.Millisecond * 25)
	}
}

yevsia, 2022-09-16:
`package main

import (
	"machine"
	"time"
)

func main() {

	err := machine.Timer2.Configure(machine.PWMConfig{
		Period: 16384e3, // 16.384ms
	})
	if err != nil {
		println("failed to configure PWM")
		return
	}

	channelA, err := machine.Timer2.Channel(machine.PD3)
	if err != nil {
		println("failed to configure channel A")
		return
	}

	duty, delta := 0, 1024

	for {

		machine.Timer2.Set(channelA, uint32(duty))
		duty += delta

		if duty < 0 || duty > 65535 {
			delta *= -1
			duty += delta
		}

		time.Sleep(time.Millisecond * 25)
	}
}`

yevsia, 2022-09-16:
> 

```
package main

import (
	"machine"
	"time"
)

func main() {

	err := machine.Timer2.Configure(machine.PWMConfig{
		Period: 16384e3, // 16.384ms
	})
	if err != nil {
		println("failed to configure PWM")
		return
	}

	channelA, err := machine.Timer2.Channel(machine.PD3)
	if err != nil {
		println("failed to configure channel A")
		return
	}

	duty, delta := 0, 1024

	for {

		machine.Timer2.Set(channelA, uint32(duty))
		duty += delta

		if duty < 0 || duty > 65535 {
			delta *= -1
			duty += delta
		}

		time.Sleep(time.Millisecond * 25)
	}
}
```

aykevl, 2022-09-19:
@yevsia that is not the code that results in the error you have shared. The error mentions `pwm` but your code does not contain this variable.

yevsia, 2022-09-19:
@aykevl 

```
package main

import (
	"machine"
	"time"
)

func main() {

	err := pwm.Configure(machine.PWMConfig{
		Period: 16384e3, // 16.384ms
	})
	if err != nil {
		println("failed to configure PWM")
		return
	}

	channelA, err := pwm.Channel(pinA)
	if err != nil {
		println("failed to configure channel A")
		return
	}

	duty, delta := 0, 1024

	for {

		pwm.Set(channelA, uint32(duty))
		duty += delta

		if duty < 0 || duty > 65535 {
			delta *= -1
			duty += delta
		}

		time.Sleep(time.Millisecond * 25)
	}
}
```

this code generate error:
```
src\TinyGo\PWM\main2.go:12:9: undeclared name: pwm
src\TinyGo\PWM\main2.go:20:19: undeclared name: pwm
src\TinyGo\PWM\main2.go:20:31: undeclared name: pinA
src\TinyGo\PWM\main2.go:30:3: undeclared name: pwm
```

If change pwm -> machine.Timer2 and pinA -> machine.PD3. (https://github.com/tinygo-org/tinygo/blob/release/src/examples/pwm/arduino.go), the code compiles but does not work correctly.

aykevl, 2022-09-29:
Please take a look at the tutorial: https://tinygo.org/docs/tutorials/pwm/

It describes how to use the `.Set()` function:

```go
pwm.Set(ch, pwm.Top()/uint32(i))
```

You should use the `Top()` method instead of relying on a fixed value.

davecheney, 2026-10-24:
Thanks for the report. This standalone example fades an LED on Arduino Uno pin D3. Connect D3 through a current-limiting resistor and LED to GND.

The duty cycle is scaled to `pwm.Top()` instead of assuming a maximum of 65535.

```go
package main

import (
	"machine"
	"time"
)

func main() {
	pwm := machine.Timer2
	if err := pwm.Configure(machine.PWMConfig{Period: 128e3}); err != nil {
		panic(err)
	}
	channel, err := pwm.Channel(machine.PD3)
	if err != nil {
		panic(err)
	}

	top := pwm.Top()
	percent, step := 0, 1
	for {
		pwm.Set(channel, top*uint32(percent)/100)
		time.Sleep(25 * time.Millisecond)
		if percent == 100 {
			step = -1
		} else if percent == 0 {
			step = 1
		}
		percent += step
	}
}
```

Save as `main.go` and run:

```sh
tinygo flash -target=arduino main.go
```

This compiles for `arduino`; I have not verified it on hardware.

closed 2026-10-24 by davecheney