tinygo-org/tinygo #3126: Unable to get output from Raspberry Pi Pico on 0.25.0

Arsen6331, 2022-09-02:
In TinyGo 0.24.0, I used UART0 with a UART to USB adapter to get output on my computer. Once I updated to 0.25.0, that stopped working.

I looked in the changelog and saw that RP2040 now uses USB as the default serial device, so I tried using `/dev/ttyACM0`, which is the serial device that appears when I plug in my pico, and there was no output on that. No other serial ports appear.

I then tried to set `machine.Serial` to `machine.UART0`, but it doesn't implement `machine.Serialer`, so that didn't work either.

Currently, I've downgraded to 0.24.0 to see the output. How do I get the serial output in 0.25.0?

sago35, 2022-09-02:
@Arsen6331 
What does the target use?
What are the results of tinygo version?

Is the following a solution?

```
fmt.Fprintf(machine.UART0, "hello world\r\n")
```

Arsen6331, 2022-09-02:
>  What does the target use?

I don't know what you mean by this. I am using the `pico` target.

>  What are the results of tinygo version?

Before update: tinygo version 0.24.0 linux/amd64 (using go version go1.19 and LLVM version 14.0.0)
After update: tinygo version 0.25.0 linux/amd64 (using go version go1.19 and LLVM version 14.0.0)

Is the following a solution?

Yes, but I'd like `machine.Serial` to point to the output, as I have many `log.Println()` and `println()` calls in my code, and I do not want to have to modify each one if possible.

sago35, 2022-09-02:
Do the following work?
It should output to `/dev/ttyACM0`, etc.

```
tinygo flash --target pico examples/serial
```

And the following should be equivalent output to the previous version.
It should output to `machine.DefaultUART` (`machine.UART0`) .

```
tinygo flash --target pico --serial uart examples/serial
```

Arsen6331, 2022-09-02:
I'm using nonstandard UART pins, so I had to make a simple program instead of using `examples/serial`. Here is what I used:

```go
package main

import (
	"io"
	"machine"
	"time"
)

func main(){
	err := machine.UART0.Configure(machine.UARTConfig{
		BaudRate: 115200,
		TX:       machine.GP12,
		RX:       machine.GP13,
	})
	if err != nil {
		panic(err)
	}
		
	for {
		io.WriteString(machine.DefaultUART, "Hello World")
		time.Sleep(50*time.Millisecond)
	}
}
```

This program worked on 0.24.0, and 0.25.0 with `--serial uart`, but I get no output on ttyACM0 with your first example on 0.25.0,

Arsen6331, 2022-09-02:
I can use `--serial uart` for now, but it is interesting why USB output isn't working.

sago35, 2022-09-03:
I have checked on Windows and Ubuntu20.04 and both work.

Also try the following for USBCDC (/dev/ttyACMx).

* Try this with the device connected to the USB Hub unplugged
* Stop using the USB Hub and connect directly to the computer.
* Check `dmesg`

Perhaps it is due to errata RP2040-E5.
https://github.com/tinygo-org/tinygo/pull/2991

Arsen6331, 2022-09-03:
It could be the USB hub actually, I'll try connecting it directly to the computer and see if it works.

Arsen6331, 2022-09-03:
It still doesn't work even if directly connected to the computer. I checked dmesg, and I seem to be having the same problem as #3012 

```
[38414.028032] usb 7-2: new full-speed USB device number 117 using xhci_hcd
[38414.162792] usb 7-2: device descriptor read/all, error -71
[38414.276029] usb 7-2: new full-speed USB device number 118 using xhci_hcd
[38414.411785] usb 7-2: device descriptor read/all, error -71
[38414.411833] usb usb7-port2: attempt power cycle
[38414.791020] usb 7-2: new full-speed USB device number 119 using xhci_hcd
[38414.810790] usb 7-2: device descriptor read/8, error -71
[38414.934790] usb 7-2: device descriptor read/8, error -71
[38415.158017] usb 7-2: new full-speed USB device number 120 using xhci_hcd
[38415.177785] usb 7-2: device descriptor read/8, error -71
[38415.302796] usb 7-2: device descriptor read/8, error -71
[38415.411082] usb usb7-port2: unable to enumerate USB device
```

aykevl, 2022-09-15:
I think I see where it goes wrong.

In your code sample:

```go
package main

import (
	"io"
	"machine"
	"time"
)

func main(){
	err := machine.UART0.Configure(machine.UARTConfig{
		BaudRate: 115200,
		TX:       machine.GP12,
		RX:       machine.GP13,
	})
	if err != nil {
		panic(err)
	}
		
	for {
		io.WriteString(machine.Serial, "Hello World")
		time.Sleep(50*time.Millisecond)
	}
}
```

You configure `machine.UART0`, but then you write to `machine.Serial`. The `machine.Serial` output is for the standard UART output (configured with the `-serial=` flag). So if you don't use `-serial=uart`, then the code won't work because it point to a different serial output.

I recommend changing the code like this:

```go
		io.WriteString(machine.UART0, "Hello World")
```

This way, you explicitly write to UART0. It will work regardless of the `-serial=` flag.

> I can use `--serial uart` for now, but it is interesting why USB output isn't working.

With `-serial=uart`, _all_ output (like println, panic) is written to the UART output. If you want println/panic/os.Stdout/etc to write to the USB-CDC output, then you should not use `-serial=uart`.

In summary: with the `-serial=` flag, you can control `machine.Serial`, which in turn affects `println`/`panic`/`os.Stdout` etc. If you want to use UART separate from USB-CDC, you can do so by configuring `machine.UART0` and writing to `machine.UART0` as a `io.Writer`.

Arsen6331, 2022-09-15:
> This way, you explicitly write to UART0. It will work regardless of the -serial= flag.

I know. That's what I am doing now. The problem was I didn't notice the `--serial` flag in the help text, so I thought that serial was broken by the update. I now know that is not the case and am using `UART0` instead.

I was just interested to know why USB-CDC wasn't working _without_ the `--serial=uart` flag. It isn't critical to my use case since I am using UART, but I am wondering why it's not working

aykevl, 2022-09-15:
> I was just interested to know why USB-CDC wasn't working _without_ the `--serial=uart` flag. It isn't critical to my use case since I am using UART, but I am wondering why it's not working

Huh, that's odd. Yes, that could be a bug.

bluntelk, 2022-09-16:
Hi There,

Not sure if this is the same. On the pico i have a simple program that println's and blinks the LED

When I plug it in I get this in dmesg (and no addition /dev/ttyACM<N> - I already have an existing ttyACM0)

```[37017.224673] usb 3-2.1: new full-speed USB device number 63 using xhci_hcd
[37017.461847] usb 3-2.1: device descriptor read/all, error -32
[37017.541158] usb 3-2.1: new full-speed USB device number 64 using xhci_hcd
[37017.778380] usb 3-2.1: device descriptor read/all, error -32
[37017.779058] usb 3-2-port1: attempt power cycle
[37018.380685] usb 3-2.1: new full-speed USB device number 65 using xhci_hcd
[37018.513848] usb 3-2.1: device descriptor read/8, error -32
[37018.720935] usb 3-2.1: device descriptor read/8, error -32
[37018.908708] usb 3-2.1: new full-speed USB device number 66 using xhci_hcd
[37019.041285] usb 3-2.1: device descriptor read/8, error -32
[37019.252083] usb 3-2.1: device descriptor read/8, error -32
[37019.357022] usb 3-2-port1: unable to enumerate USB device
[37019.436578] usb 3-2.4: new full-speed USB device number 67 using xhci_hcd
[37019.541100] usb 3-2.4: device descriptor read/64, error -32
[37019.728625] usb 3-2.4: device descriptor read/64, error -32
[37019.917181] usb 3-2.4: new full-speed USB device number 68 using xhci_hcd
[37020.025134] usb 3-2.4: device descriptor read/64, error -32
[37020.213061] usb 3-2.4: device descriptor read/64, error -32
[37020.321262] usb 3-2-port4: attempt power cycle
[37020.924700] usb 3-2.4: new full-speed USB device number 69 using xhci_hcd
[37020.954095] usb 3-2.4: Device not responding to setup address.
[37021.189019] usb 3-2.4: Device not responding to setup address.
[37021.396793] usb 3-2.4: device not accepting address 69, error -71
[37021.476597] usb 3-2.4: new full-speed USB device number 70 using xhci_hcd
[37021.505042] usb 3-2.4: Device not responding to setup address.
[37021.741064] usb 3-2.4: Device not responding to setup address.
[37021.948594] usb 3-2.4: device not accepting address 70, error -71
[37021.949045] usb 3-2-port4: unable to enumerate USB device```

alvarolm, 2022-10-15:
> Hi There,
> 
> Not sure if this is the same. On the pico i have a simple program that println's and blinks the LED
> 
> When I plug it in I get this in dmesg (and no addition /dev/ttyACM - I already have an existing ttyACM0)
> 
> ```
> [37017.461847] usb 3-2.1: device descriptor read/all, error -32
> [37017.541158] usb 3-2.1: new full-speed USB device number 64 using xhci_hcd
> [37017.778380] usb 3-2.1: device descriptor read/all, error -32
> [37017.779058] usb 3-2-port1: attempt power cycle
> [37018.380685] usb 3-2.1: new full-speed USB device number 65 using xhci_hcd
> [37018.513848] usb 3-2.1: device descriptor read/8, error -32
> [37018.720935] usb 3-2.1: device descriptor read/8, error -32
> [37018.908708] usb 3-2.1: new full-speed USB device number 66 using xhci_hcd
> [37019.041285] usb 3-2.1: device descriptor read/8, error -32
> [37019.252083] usb 3-2.1: device descriptor read/8, error -32
> [37019.357022] usb 3-2-port1: unable to enumerate USB device
> [37019.436578] usb 3-2.4: new full-speed USB device number 67 using xhci_hcd
> [37019.541100] usb 3-2.4: device descriptor read/64, error -32
> [37019.728625] usb 3-2.4: device descriptor read/64, error -32
> [37019.917181] usb 3-2.4: new full-speed USB device number 68 using xhci_hcd
> [37020.025134] usb 3-2.4: device descriptor read/64, error -32
> [37020.213061] usb 3-2.4: device descriptor read/64, error -32
> [37020.321262] usb 3-2-port4: attempt power cycle
> [37020.924700] usb 3-2.4: new full-speed USB device number 69 using xhci_hcd
> [37020.954095] usb 3-2.4: Device not responding to setup address.
> [37021.189019] usb 3-2.4: Device not responding to setup address.
> [37021.396793] usb 3-2.4: device not accepting address 69, error -71
> [37021.476597] usb 3-2.4: new full-speed USB device number 70 using xhci_hcd
> [37021.505042] usb 3-2.4: Device not responding to setup address.
> [37021.741064] usb 3-2.4: Device not responding to setup address.
> [37021.948594] usb 3-2.4: device not accepting address 70, error -71
> [37021.949045] usb 3-2-port4: unable to enumerate USB device```
> ```

had exactly the same issue, resolved after changing the cable and rebooting as mass storage mode (bootsel), no idea why.

kenbell, 2022-10-20:
It looks like everything has been resolved in this issue.  Any objections to closing?

deadprogram, 2024-11-30:
Now closing, thanks everyone!

closed 2024-11-30 by deadprogram