tinygo-org/tinygo #3147: support //go:wasmimport or a similarly named alias for imported functions

codefromthecrypt, 2022-09-09:
Let's support `//go:wasmimport` or a similarly named alias for imported functions. This would be an alias to `//export`, so that existing code can work, but much more intuitive. By being more intuitive, new users have a better experience, and we have less load on issues as well less angst on support.

## Background

Currently, users need to add the keyword `//export` in order to import a function (ex from the wasm host).

ex.
```
//go:wasm-module env
//export log
func _log(ptr uint32, size uint32)
```

This is very unintuitive and provides a bad first time experience for TinyGo users. It also occupies many issues and in doing so steals time from folks contributing support. A couple examples are #3000 and #455, but there are many more occurrences than this, including my personal first experience.

As far as I understand the history from @aykevl, 
> "Export" in this case really means "make available to C" - either as a declaration or a definition.

I would argue that this can be true, but jargon that doesn't match is also more confusing than re-use of some C related jargon in Go. Many Go developers won't think through like this, hence the continual support questions.

More background could be that Go didn't have an `import` keyword. For example, In Go, you need to define a .s file then use a `CallImport` instruction there. The point remains that it also uses the word import not export for this.
https://github.com/golang/go/blob/master/src/syscall/js/js_js.s#L63-L65

However, if you look at the latest on Go, https://github.com/golang/go/issues/38248, it appears `//go:wasmimport` would be used, not `//export`, for importing functions.

fgsch, 2022-09-09:
fwiw, TinyGo also supports `//go:export`, which is what `//export` is rewritten to.

james-lawrence, 2022-09-10:
minor note that a [guide](https://tinygo.org/docs/guides/webassembly/) should include a end to end example of a module exporting a function, another module importing it, and then executing the resulting modules. something along [this line](https://github.com/james-lawrence/tinygo-wasi-bug).

codefromthecrypt, 2022-09-11:
@james-lawrence forked an issue for your above comment so it doesn't end up lost. This example can be made now and syntax changed later (among the other ones) https://github.com/tinygo-org/tinygo/issues/3154

aykevl, 2022-09-15:
I missed the fact that `//go:wasmimport` is now accepted upstream. This means we're free to use it.
I made a PR to add support for it: https://github.com/tinygo-org/tinygo/pull/3165

Long term, I'd also like to change the behavior of `//export` so it isn't a wasm export, and instead use a `//go:wasmexport` or something similar. But there is a much greater risk at incompatibility with that with the upstream Go toolchain, so I'm not entirely sure about it. (Of course, there will be a transition period for such a rather big change).

deadprogram, 2023-06-14:
This was completed as part of release `v0.28` so now closing. Thanks!

closed 2023-06-14 by deadprogram