tinygo-org/tinygo #3150: init() works differently in standard-Go and tinygo

sago35, 2022-09-09:
For the following source code, standard-Go and tinygo work differently.
However, I do not see much of a problem.

`main.go`
```go
package main

import "fmt"

var (
	msg string
)

func main() {
	fmt.Printf("msg : %q\n", msg)
}
```

`override.go`
```go
package main

func init() {
	msg = "override"
}
```

In Go, `init()` is called correctly in all ways.

```
$ go run .
msg : "override"

$ go run main.go override.go
msg : "override"

$ go run override.go main.go
msg : "override"
```

TinyGo does not work well except with folder specified.

```
$ tinygo run .
msg : "override"

$ tinygo run main.go override.go
msg : ""

$ tinygo run override.go main.go
# command-line-arguments
override.go:4:2: undeclared name: msg

```

aykevl, 2022-09-14:
This is not a bug in `init` but rather in the way command line parameters are parsed. Therefore, I've changed the bug title.

> ```
> $ tinygo run main.go override.go
> msg : ""
> ```

This runs the file main.go with the command line argument `override.go`. It is similar to the following:

```
tinygo build -o main main.go
./main override.go
```

Same here:

> ```
> $ tinygo run override.go main.go
> # command-line-arguments
> override.go:4:2: undeclared name: msg
> ```

TinyGo tries to compile just the file override.go which of course won't work.

davecheney, 2026-10-24:
Duplicate of #1150

Thanks for reporting this. `go run` accepts multiple source files from the same package when they form a complete program. TinyGo currently treats the second filename as a program argument instead.

A small reproducer:

`main.go`:
```go
package main

func main() {
    println(message)
}
```

`message.go`:
```go
package main

const message = "hello from both files"
```

With Go 1.27.3:
```text
$ go run main.go message.go
hello from both files
```

With TinyGo 0.44.0-dev-eae3f875:
```text
$ tinygo run main.go message.go
# command-line-arguments
main.go:4:10: undefined: message
```

Running the package with `tinygo run .` works and prints `hello from both files`.

Closing this as a duplicate so we can track multiple-source-file support in #1150. This remains a TinyGo limitation, not a problem with the program.

davecheney, 2026-10-24:
Duplicate of #1150

closed 2026-10-24 by davecheney