tinygo-org/tinygo #3151: cgo build error - undefined symbol fcntl

dunrix, 2022-09-10:
Attempt to write a simple call to (musl?) C function, fails to compile at linking stage.
TinyGo 0.24.0
Linux x86_64 

```go
package main

/*
#include <clang-c/Index.h>
#include <llvm/Config/llvm-config.h>
#include <stdio.h>
*/
import "C"

func main() {
        // spawn `ls` command - list directory contents
	var cmd = C.CString("ls")
	prc := C.popen(cmd, C.CString("r"))
	C.pclose(prc)

	print(C.LLVM_VERSION_STRING, "\n")
}
```

tinygo build results in following error
> tinygo:ld.lld: error: undefined symbol: fcntl
>>> referenced by popen.c:42 (/usr/lib/tinygo/lib/musl/src/stdio/popen.c:42)
>>>               popen.c.o:(popen) in archive /home/user/.cache/tinygo/musl-x86_64-unknown-linux-x86-64/lib.a
>>> referenced by popen.c:59 (/usr/lib/tinygo/lib/musl/src/stdio/popen.c:59)
>>>               popen.c.o:(popen) in archive /home/user/.cache/tinygo/musl-x86_64-unknown-linux-x86-64/lib.a

Is it just not yet implemented or is there some other way how to link the C lib?

aykevl, 2022-09-14:
It is not implemented. Probably the only thing needed is to include some more files in https://github.com/tinygo-org/tinygo/blob/release/builder/musl.go#L108.

dunrix, 2022-09-15:
@aykevl Thanks for the reply and yes, I'd like to spawn an external command and catch its output. Adapt original Go program which uses `os/exec`.

davecheney, 2026-10-28:
The missing `fcntl` linker error no longer reproduces. I built this example with TinyGo `0.44.0-dev-eae3f875` for Linux/AMD64 and ran it on a Linux host:

```go
package main

/*
#include <stdio.h>
#include <stdlib.h>
*/
import "C"
import "unsafe"

func main() {
	cmd := C.CString("printf tinygo-popen-ok")
	mode := C.CString("r")
	defer C.free(unsafe.Pointer(cmd))
	defer C.free(unsafe.Pointer(mode))

	p := C.popen(cmd, mode)
	if p == nil {
		panic("popen failed")
	}
	var buf [64]C.char
	line := C.fgets(&buf[0], C.int(len(buf)), p)
	status := C.pclose(p)
	if line == nil || C.GoString(&buf[0]) != "tinygo-popen-ok" || status != 0 {
		panic("unexpected command output or exit status")
	}
	println("popen output and exit status verified")
}
```

```text
$ GOOS=linux GOARCH=amd64 tinygo build -o popen-test main.go
$ ./popen-test
popen output and exit status verified
```

Closing as fixed. Thanks for reporting it.

closed 2026-10-28 by davecheney