tinygo-org/tinygo #3157: tinygo MIDI community

davidhtims, 2022-09-12:
Where do the tinygo USB-MIDI people hang out please?
I have little experience with microcontrollers and I'm looking for a board to start experimenting with MIDI data manipulation.

deadprogram, 2022-09-15:
There is not a specific place (yet) for TinyGo MIDI people, just the Slack channel.

I am really happy you are interested in this! :musical_keyboard:

davidhtims, 2022-10-16:
I have started a Google Group:
[https://groups.google.com/g/tinygomidi](https://groups.google.com/g/tinygomidi)

At the moment anyone can join without asking ...

deadprogram, 2023-09-03:
I have just added https://github.com/tinygo-org/tinygo/pull/3889 to improve on the current MIDI support. Please take a look!

davecheney, 2026-10-26:
Thanks for sharing the group. I’m closing this as the question has been answered. Please open a new issue if you have any further questions about using TinyGo.

closed 2026-10-26 by davecheney