tinygo-org/tinygo #3171: USB-HID: Media/Consumer keys is not work on nrf52840

labrusca, 2022-09-16:
Seem Media/Consumer keys is not working on my M60 Keyboard(base on nrf52840),but the other keys is working well.
system: Windows 11

aykevl, 2022-09-19:
What has this to do with TinyGo? Are you programming it using TinyGo?

labrusca, 2022-09-19:
> What has this to do with TinyGo? Are you programming it using TinyGo?

I tried to write firmware for my mechanical keyboard using tinygo and it works fine so far except for the Media/Consumer keys, which don't work

sago35, 2022-09-19:
Yes, Media/Consumer keys are not currently supported.

https://github.com/tinygo-org/tinygo/blob/v0.25.0/src/machine/usb/hid/keyboard/keyboard.go#L233

It was in the source code we based it on, but we have not been able to implement it so far.

https://github.com/ardnew/tinygo/blob/6dcabd49800981e31c94d0ce5a571376c20b854b/src/machine/usb/dhw-hid_atsamd51.go#L126

labrusca, 2022-09-19:
@sago35 
Oh ,I see,thanks.thanks. Maybe I'll drop this feature for now until it's implemented.

neildavis, 2023-02-07:
+1 for implementing USB Consumer Control events (e.g. volume). It's the only thing currently stopping me from using TinyGo instead of CircuitPython on my USB input boards :)

Is something blocking this or is it just a case of priority/time?

sago35, 2023-02-09:
@labrusca @neildavis 
If you are interested, check out the following

#3436

deadprogram, 2023-06-14:
This was completed as part of release `v0.28` so now closing. Thanks!

closed 2023-06-14 by deadprogram