tinygo-org/tinygo #3182: About memory management tinygo

dedalqq, 2022-09-18:
I have a few questions and suggestions about how the garbage collector works.

- Would it be a good idea to implement manual memory clearing `runtime.Free(interface{})`?
- How about implementing a reference-counted garbage collector?

I think reference count based memory management is very good. Of course, leaks are possible with cyclic references, but this is extremely rarely needed.

I also think that giving the opportunity to manually clear the memory is also good. This should not interfere with the work of the garbage collector, but it can reduce the number of garbage collector cycles if not completely removed.

What do you think about all this?

Thanks a lot. =)

aykevl, 2022-09-19:
Thank you for these suggestions!

> Would it be a good idea to implement manual memory clearing `runtime.Free(interface{})`?

This has come up a few times in the past, but it is quite difficult to cleanly implement this in Go. For example:

 - `interface{}` can itself be a heap object, therefore calling `runtime.Free` can actually cause a heap allocation. The way to avoid this would be to only use pointers and reference types.
 - What exactly should it deallocate? For container objects like maps and channels, should it also deallocate the values stored in there (for example, if it stores pointer values) or only the map/channel itself? What about structs that contain pointers?
 - How do you even know a value is heap allocated? The compiler decides how values are allocated (even `new(int)` could be stack allocated, or not allocated at all if the compiler determines that's safe). This makes a manual free function hard to use correctly.
 - This should be obvious, but a `runtime.Free` is unsafe. It results in a high risk for dangling pointers if not used correctly.

So in summary I do not think a manual free function is a good fit for the Go language and is certainly hard to use correctly.

> How about implementing a reference-counted garbage collector?

This is something I want to do some day. It's not as simple as in other languages, because Go allows interior pointers. This means that if you take a pointer to a struct field and call a function with that pointer as parameter, it should still increment/decrement the reference count of the struct. I think this can be implemented using fat pointers (one pointer as usual, and another pointer points to the reference count) which will require large changes to the compiler and runtime. That cost may be worth it though. I'm not working on it now because it has a lower priority to me than many other things in TinyGo.

dedalqq, 2022-09-21:
- about memory management using the reference counter:

This is great news, thanks.

- About the free function

1. I think it's not important
2 and 3. This function is supposed to recursively deallocate everything if the passed object is on the heap and do nothing if it is on the stack. It is also assumed that the developer fully understands what he is doing as well as all the problems that this can cause.
4. yes, of course =)

It's just a suggestion, I think it might be useful in some cases, but I'm not sure myself if it's really good idea but this can be useful in small programs where, with the help of debugging, you could find memory allocations and clean them up with this function if this really need to.

dedalqq, 2022-09-21:
I am closing this issue as I received answers to all questions. But if you have something to add, I will be glad.

closed 2022-09-21 by dedalqq