tinygo-org/tinygo #3186: Add CLI flag for gcDebug

anuraaga, 2022-09-20:
I noticed there is a `gcDebug` constant in the source code for printing some GC information. Would it make sense to expose it as a CLI flag to not need a custom build to debug GC? Alternatively, could we at least change from `const` to `var` to make it easier to create a custom build with `ldflags` instead of having to edit source?

aykevl, 2022-09-20:
Related to https://github.com/tinygo-org/tinygo/issues/3138

I think it's useful to enable `gcAsserts` with a build tags. I'm not so sure it's useful to do the same for `gcDebug`: this constant is primarily intended for developers of the GC.

> to not need a custom build to debug GC? 

Can you give a bit more context what you are trying to do?

anuraaga, 2022-09-21:
I'd like to have some debugging of how often GC is being performed to analyze app performance. For `gcDebug`, specifically this logging would be useful

https://github.com/tinygo-org/tinygo/blob/release/src/runtime/gc_conservative.go#L246

There are other log statements that aren't so much but just when debugging performance it would be fine to have some extra logs while being to understand the GC cycles better.

dgryski, 2022-09-21:
https://github.com/tinygo-org/tinygo/pull/3191 PR for gc assertions

Some of the stats you're looking for might be exposed in the runtime.MemStats struct; but note that 1) it's expensive to call and 2) gcDebug prints out *way* to much information to be useful for the average user of tinygo.

aykevl, 2022-09-29:
Yeah, I would suggest trying `runtime.ReadMemStats` to see whether it provides enough information. You can look at `GCSys` to know the size of the heap without metadata, and therefore see whether it has grown since the last time you called it.

dgryski, 2022-09-29:
Some of the fields in `runtime.MemoryStats` are currently stubbed out; if you need specific fields for debugging, the tracking issue is here: https://github.com/tinygo-org/tinygo/issues/3167

anuraaga, 2022-09-30:
Thanks I will take a look at the stats and possibly follow up on #3167 if anything seems missing.

closed 2022-09-30 by anuraaga