tinygo-org/tinygo #3187: net.Pipe not supported

dkegel-fastly, 2022-09-20:
Looking at https://github.com/go-jose/go-jose 
#2660 covers go-jose's need for reflect.

The next problem I noticed was a test needs net.Pipe:
```
$ cd json
$ tinygo test
# github.com/go-jose/go-jose/v3/json
stream_test.go:185:15: Pipe not declared by package net
```

johnkeates, 2023-02-01:
I did a very quick grep to see if net.Pipe is actually used directly by go-jose, but it only appears to be part of the test. Did you test other features beyond this? Or is this more of an explore-and-implement roadmap.

dkegel-fastly, 2023-02-01:
It was the barest of "what is the next low hanging fruit" kind of exploration.

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing. Thank you!

PR #3227 (2022-10-18): net: implement Pipe
Fixes: #3187 

#### Testing

This pulls in tests from official go implementation along with some helper code for the test.
closed 2023-02-12 by deadprogram