tinygo-org/tinygo #3192: Unable to install tinygo on Arm64

byteshiva, 2022-09-22:
I'm unable to install tinygo on Arm64

Error: 

```
sudo dpkg -i tinygo_0.25.0_armhf.deb 
dpkg: error processing archive tinygo_0.25.0_armhf.deb (--install):
 package architecture (armhf) does not match system (arm64)
Errors were encountered while processing:
 tinygo_0.25.0_armhf.deb
```

aykevl, 2022-09-22:
armhf is 32-bit ARM. You will need the 64-bit package:  tinygo0.25.0.linux-arm64.tar.gz

byteshiva, 2022-09-23:
> armhf is 32-bit ARM. You will need the 64-bit package: tinygo0.25.0.linux-arm64.tar.gz

Thanks it worked.

```
$ wget https://github.com/tinygo-org/tinygo/releases/download/v0.25.0/tinygo_0.25.0_arm64.deb

$ sudo dpkg -i tinygo_0.25.0_arm64.deb 
Selecting previously unselected package tinygo.
(Reading database ... 29936 files and directories currently installed.)
Preparing to unpack tinygo_0.25.0_arm64.deb ...
Unpacking tinygo (0.25.0) ...
Setting up tinygo (0.25.0) ...

$ tinygo 
No command-line arguments supplied.
TinyGo is a Go compiler for small places.
version: 0.25.0
usage: tinygo <command> [arguments]

commands:
  build:   compile packages and dependencies
  run:     compile and run immediately
  test:    test packages
  flash:   compile and flash to the device
  gdb:     run/flash and immediately enter GDB
  lldb:    run/flash and immediately enter LLDB
  env:     list environment variables used during build
  list:    run go list using the TinyGo root
  clean:   empty cache directory (/root/.cache/tinygo)
  targets: list targets
  info:    show info for specified target
  version: show version
  help:    print this help text

for more details, see https://tinygo.org/docs/reference/usage/
$ 
```

closed 2022-09-23 by aykevl