tinygo-org/tinygo #3197: WASM: MakeMapWithSize not declared by package reflect

gedw99, 2022-09-26:
https://github.com/henrixapp/pdfcomprezzor

```

cd pdfcomprezzor && GOOS=js GOARCH=wasm go build -o app.wasm .
go: downloading github.com/pdfcpu/pdfcpu v0.3.13
cd pdfcomprezzor && tinygo build -o app-tiny.wasm -target wasm .
# encoding/gob
/usr/local/opt/go/libexec/src/encoding/gob/decode.go:562:21: MakeMapWithSize not declared by package reflect

````

dgryski, 2022-09-28:
This is a known limitation of the current `reflect` package.

gedw99, 2022-10-06:
thanks @dgryski for feedback.

leonbotros, 2022-12-02:
Running into this as well.

aykevl, 2022-12-12:
An easy fix would be to stub it out, like here:

https://github.com/tinygo-org/tinygo/blob/b5ad81c884981f42c1ac491cc1d195aea886ddbe/src/reflect/value.go#L881-L884

Feel free to send a PR :)

dgryski, 2023-02-24:
My recent reflect work #3470 implements this call.

deadprogram, 2023-06-14:
This was completed as part of release `v0.28` so now closing. Thanks!

closed 2023-06-14 by deadprogram