tinygo-org/tinygo #3199: RP2020 RingBuffer limitation?

neacon-ivalkenburg, 2022-09-26:
Hey,

im currently using TinyGO along with a raspberry pico to talk to a sim7020e NB-IoT board. The problem im having is that one of the MQTT related AT command returns a rather long reply (+CMQPUB). This reply when read the buffer with UART.Read() gets only read for about ~130 characters. I've tried UART.ReadByte() in quick loop, but the results are somewhat unreliable; only 1 out of 10 i get the full reply.

So im wondering if the RingBuffer (set at 128 bytes) is causing this? And if there is anything i could do or someone could point me in the right direction.

Thanks

aykevl, 2022-09-29:
Yes, the ringbuffer size is currently not configurable:

https://github.com/tinygo-org/tinygo/blob/b5ad81c884981f42c1ac491cc1d195aea886ddbe/src/machine/buffer.go#L7

This is unfortunately not easy to fix without increasing the code size or memory consumption in the common case. But I agree this needs to be fixed somehow.

soypat, 2022-10-04:
Hmm, a quick and dirty solution would be to add a file `tinygo/src/machine/large_buffer.go` with build tags including rp2040 (and excluding rp2040 in buffer.go). In this file you could implement a version of buffer with larger buf. 

I'd imagine there's a better solution in the long run, but I'm not familiar with how RingBuffer is used in practice. And why is such a generic type in the machine package exported? The volatiles tell me there's underlying hardware involved, but is this data structure that common across boards?