tinygo-org/tinygo #3209: ErrProcess Not declared by package os

nerdeveloper, 2022-09-29:
Hi folks

When i do 
```
tinygo build -o ./dist/app.wasm -target wasm ./...
```

I get this error
```
# os/exec
/usr/local/Cellar/tinygo/0.25.0/src/os/exec/exec.go:659:36: ErrProcessDone not declared by package os
```
 Go version: go version go1.19.1 darwin/amd64
 Tinygo version: tinygo version 0.25.0 darwin/amd64 (using go version go1.19.1 and LLVM version 14.0.0)
 
 I dont know what to do here. Kindly assist.

deadprogram, 2022-09-29:
I do not think `os.Exec` is currently supported, and if it is anywhere, is probably not on `wasm` target since sandbox etc.

aykevl, 2022-09-29:
This specific error will be fixed with the next release: https://github.com/tinygo-org/tinygo/pull/3060/commits/a8e34e825d772aeb06d35d6029b436c0d20b0f4d
However, like @deadprogram said we don't support `os.Exec` right now. (I've looked into it but supporting it is not at all easy).

nerdeveloper, 2022-09-30:
thank you folks

closed 2022-09-30 by nerdeveloper