tinygo-org/tinygo #3215: panic: runtime error: index out of range [0] with length 0

garet90, 2022-10-03:
When building my project with the following command
`GOOS=js GOARCH=wasm tinygo build -o ./bin/client.wasm ./app/client`
I get the following log
```
panic: runtime error: index out of range [0] with length 0

goroutine 62 [running]:
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionStart(0xc0049852c0, 0x0)
        D:/a/tinygo/tinygo/compiler/compiler.go:1119 +0x11c5
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc0049852c0)
        D:/a/tinygo/tinygo/compiler/compiler.go:1201 +0x38
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0xc002846000, 0xc0037ac9c0)
        D:/a/tinygo/tinygo/compiler/symbol.go:199 +0xbbc
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall(0xc004985180, 0xc0049c21c0)
        D:/a/tinygo/tinygo/compiler/compiler.go:1687 +0x826
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0xc004985180, {0x7ff663235be8?, 0xc0049c2180?})
        D:/a/tinygo/tinygo/compiler/compiler.go:1841 +0x2573
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc004985180, {0x7ff663235ba0?, 0xc0049c2180?})
        D:/a/tinygo/tinygo/compiler/compiler.go:1344 +0x955
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc004985180)
        D:/a/tinygo/tinygo/compiler/compiler.go:1242 +0x9e9
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0xc002846000, 0xc003c40b60)
        D:/a/tinygo/tinygo/compiler/symbol.go:199 +0xbbc
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall(0xc004985040, 0xc00499f240)
        D:/a/tinygo/tinygo/compiler/compiler.go:1687 +0x826
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0xc004985040, {0x7ff663235be8?, 0xc00499f200?})
        D:/a/tinygo/tinygo/compiler/compiler.go:1841 +0x2573
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc004985040, {0x7ff663235ba0?, 0xc00499f200?})
        D:/a/tinygo/tinygo/compiler/compiler.go:1344 +0x955
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc004985040)
        D:/a/tinygo/tinygo/compiler/compiler.go:1242 +0x9e9
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0xc002846000, {0xc000539520?}, 0xc000db9600)
        D:/a/tinygo/tinygo/compiler/compiler.go:843 +0xa0f
github.com/tinygo-org/tinygo/compiler.CompilePackage({0xc0002d5e00?, 0xc00093e210?}, 0xc000323c20, 0xc000db9600, {0x1006637303836?}, 0x7ff66319c95d?, 0x0?)
        D:/a/tinygo/tinygo/compiler/compiler.go:290 +0x417
github.com/tinygo-org/tinygo/builder.Build.func3(0xc00358d0e0)
        D:/a/tinygo/tinygo/builder/build.go:354 +0x21a
github.com/tinygo-org/tinygo/builder.runJob(0xc00358d0e0, 0x0?)
        D:/a/tinygo/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
        D:/a/tinygo/tinygo/builder/jobs.go:123 +0x5db
```

output of tinygo info:
```
LLVM triple:       x86_64-unknown-windows-gnu
GOOS:              windows
GOARCH:            amd64
build tags:        windows amd64 tinygo math_big_pure_go gc.conservative scheduler.tasks serial.none
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     C:\Users\ghall\AppData\Local\tinygo\goroot-b51d0006217f7e788a0799711e87660c009e82aa5fa545dd8daf07c292d845f1
```

Running with the latest commit (as of 10/3/2022)
The project is fairly big so I do not have a MRE

garet90, 2022-10-03:
The only dependency of the project is golang.org/x/exp/contraints and there is no cgo as well

garet90, 2022-10-03:
placing an equal sign between -o and the output made it work

closed 2022-10-03 by garet90

zenhack, 2022-11-21:
I hit this as well, and using `-o=...` worked for me too -- but maybe this should be reopened; a panic doesn't seem like reasonable behavior here.

aykevl, 2022-12-08:
>  The project is fairly big so I do not have a MRE

Then we can't fix it. So if you want this fixed, please share a reproducer.
In any case, this looks like a real bug.

vlkv, 2023-02-21:
Hello, 
I am hitting this **randomly** from time to time while I work on my project in a manner: wrote some code...started tests with `tinygo test -v ./...`

```
$ tinygo version
tinygo version 0.26.0 linux/amd64 (using go version go1.19.1 and LLVM version 14.0.0)
$ cd ./junk/go_indie
$ tinygo test -v ./...
go: downloading golang.org/x/exp v0.0.0-20230212135524-a684f[29](https://gitlab.takeprofit.team/backend/indicators/tpi_core/-/jobs/52520#L29)349b6
panic: runtime error: index out of range [0] with length 0
goroutine 1191 [running]:
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionStart(0xc007ff5400, 0x0)
	/__w/tinygo/tinygo/compiler/compiler.go:1119 +0x11c5
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc007ff5400)
	/__w/tinygo/tinygo/compiler/compiler.go:1201 +0x38
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0xc00498c780, 0xc00ca94d00)
	/__w/tinygo/tinygo/compiler/symbol.go:199 +0xbbc
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall(0xc007ff5180, 0xc0085d2ec0)
	/__w/tinygo/tinygo/compiler/compiler.go:1687 +0x826
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0xc007ff5180, {0x52e6028?, 0xc0085d2e80?})
	/__w/tinygo/tinygo/compiler/compiler.go:1841 +0x2573
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc007ff5180, {0x52e5fe0?, 0xc0085d2e80?})
	/__w/tinygo/tinygo/compiler/compiler.go:1[34](https://gitlab.takeprofit.team/backend/indicators/tpi_core/-/jobs/52520#L34)4 +0x955
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc007ff5180)
	/__w/tinygo/tinygo/compiler/compiler.go:1242 +0x9e9
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0xc00498c780, {0xc00c59e340?}, 0xc005a5b900)
	/__w/tinygo/tinygo/compiler/compiler.go:843 +0xa0f
github.com/tinygo-org/tinygo/compiler.CompilePackage({0xc002074460?, 0xc0090de4e0?}, 0xc001947a40, 0xc005a5b900, {0xc009450b80?}, 0x525c244?, 0x49?)
	/__w/tinygo/tinygo/compiler/compiler.go:290 +0x417
github.com/tinygo-org/tinygo/builder.Build.func3(0xc00c7f0780)
	/__w/tinygo/tinygo/builder/build.go:[35](https://gitlab.takeprofit.team/backend/indicators/tpi_core/-/jobs/52520#L35)4 +0x216
github.com/tinygo-org/tinygo/builder.runJob(0xc00c7f0780, 0xc00005e7b8?)
	/__w/tinygo/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
	/__w/tinygo/tinygo/builder/jobs.go:123 +0x5db
Cleaning up project directory and file based variables
00:00
ERROR: Job failed: exit code 1
```

1) Where should I put the equal sign here?..
2) I cannot reproduce it, because it is random
3) `tinygo clean ./...` heals the problem

aykevl, 2023-02-21:
> 1. Where should I put the equal sign here?..

Which equal sign?

> 2. I cannot reproduce it, because it is random (looks to me so)
> 3. `tinygo clean ./...` heals the problem

Can you try and see whether you can reliably crash TinyGo by running `tinygo clean` and then `tinygo test` (or something similar)? And if you do, can you share the exact steps you took to do that (including the source code)?
Having random crashes is unusual, but of course not impossible. It makes reproducing them a bit harder though (I really need to be able to reproduce the crash myself to be able to fix the problem).

vlkv, 2023-02-21:
> Can you try and see whether you can reliably crash TinyGo by running tinygo clean and then tinygo test (or something similar)?

I understand that you need an MRE. At the moment I don't have it unfortunately, and yes, I will definitely try to create one. Will report if I succeed with it.

gggrafff, 2023-06-07:
Hello! I have the same problem. My project is big enough, so I'll try to create MRE. 

I have added some debug output to tinygo/compiler/compiler.go:
```
	} else {
		fmt.Printf("compiler.go:1153, len(b.blockEntries)==%d, len(b.fn.Blocks)==%d, b.fn==%v \n", len(b.blockEntries), len(b.fn.Blocks), b.fn)
		for _, block := range b.fn.DomPreorder() {
			fmt.Printf("compiler.go:1154, b.ctx.AddBasicBlock \n")
			llvmBlock := b.ctx.AddBasicBlock(b.llvmFn, block.Comment)
			b.blockEntries[block] = llvmBlock
			b.blockExits[block] = llvmBlock
		}
		// Normal functions have an entry block.
		fmt.Printf("compiler.go:1159, len(b.blockEntries)==%d, len(b.fn.Blocks)==%d \n", len(b.blockEntries), len(b.fn.Blocks))
		entryBlock = b.blockEntries[b.fn.Blocks[0]]
	}
```

And tinygo's output is:
```
$ /home/ants/Projects/tinygo/build/tinygo build -o main.wasm -target ./target.json -opt=2 .
compiler.go:1153, len(b.blockEntries)==0, len(b.fn.Blocks)==1, b.fn==gitlab.company.team/backend/indicators/tpi_runner/go_runner/indicators/user_indie.init 
compiler.go:1154, b.ctx.AddBasicBlock 
compiler.go:1159, len(b.blockEntries)==1, len(b.fn.Blocks)==1 
compiler.go:1153, len(b.blockEntries)==0, len(b.fn.Blocks)==1, b.fn==(*gitlab.company.team/backend/indicators/tpi_runner/go_runner/indicators/user_indie.Main).Calc 
compiler.go:1154, b.ctx.AddBasicBlock 
compiler.go:1159, len(b.blockEntries)==1, len(b.fn.Blocks)==1 
compiler.go:1153, len(b.blockEntries)==0, len(b.fn.Blocks)==0, b.fn==(*gitlab.company.team/backend/indicators/tpi_runner/go_runner/take_profit/series.SeriesBase[float64]).GetItem[float64] 
compiler.go:1159, len(b.blockEntries)==0, len(b.fn.Blocks)==0 
panic: runtime error: index out of range [0] with length 0

goroutine 102 [running]:
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionStart(0xc002664640, 0x0)
	/home/ants/Projects/tinygo/compiler/compiler.go:1162 +0x13d7
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc002664640)
	/home/ants/Projects/tinygo/compiler/compiler.go:1251 +0x38
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0xc003e40000, 0xc003dc3680)
	/home/ants/Projects/tinygo/compiler/symbol.go:204 +0xdac
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall(0xc002664500, 0xc003fc65c0)
	/home/ants/Projects/tinygo/compiler/compiler.go:1753 +0x8a6
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0xc002664500, {0x4b8e4c8, 0xc003fc6580})
	/home/ants/Projects/tinygo/compiler/compiler.go:1909 +0x4ea
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc002664500, {0x4b8e480?, 0xc003fc6580?})
	/home/ants/Projects/tinygo/compiler/compiler.go:1394 +0x806
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc002664500)
	/home/ants/Projects/tinygo/compiler/compiler.go:1292 +0x974
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0xc003e40000, {0xc003c4e300?}, 0xc003d5a700)
	/home/ants/Projects/tinygo/compiler/compiler.go:878 +0x8ef
github.com/tinygo-org/tinygo/compiler.CompilePackage({0xc0000972c0?, 0x58?}, 0xc00030a960, 0xc003d5a700, {0xa15e3af2521484f2?}, 0x4b095e1?, 0xff?)
	/home/ants/Projects/tinygo/compiler/compiler.go:306 +0x459
github.com/tinygo-org/tinygo/builder.Build.func3(0xc003dc8ba0)
	/home/ants/Projects/tinygo/builder/build.go:357 +0x20e
github.com/tinygo-org/tinygo/builder.runJob(0xc003dc8ba0, 0x0?)
	/home/ants/Projects/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
	/home/ants/Projects/tinygo/builder/jobs.go:123 +0x5be
```

I am continuing to work on this problem.
Have you some ideas about it? What other debug data should I collect?

aykevl, 2023-06-07:
@gggrafff does it happen every time you try? Because so far, I've only seen this as an intermittent bug.

gggrafff, 2023-06-08:
@aykevl thank you
no, I get it randomly. I make shell script to run compilation in the loop:
```
#!/bin/bash
set -o xtrace  # verbose mode

for i in {1..100}
do
    /home/ants/Projects/tinygo/build/tinygo clean .

    /home/ants/Projects/tinygo/build/tinygo build -o main.wasm -target ./target.json -opt=2 .
    RES=$?
    echo $RES
    if [ $RES -ne 0 ]
    then
        echo "FAIL"
        exit 1
    fi

    /home/ants/Projects/tinygo/build/tinygo build -o main.wasm -target ./target.json -opt=2 -tags=wasmpiler .
    RES=$?
    echo $RES
    if [ $RES -ne 0 ]
    then
        echo "FAIL"
        exit 1
    fi
done
```
this script allows me to reproduce the problem on the my big project, but I haven't reproduced it yet on a small project like "hello world"