tinygo-org/tinygo #3217: rp2040: GPIO interrupt causes GC load

kenbell, 2022-10-05:
TinyGo version: `tinygo version 0.26.0 linux/amd64 (using go version go1.19 and LLVM version 14.0.0)`

Actual behaviour:
GPIO interrupts disabled for 100ms every 250ms

Expected behaviour:
GPIO interrupts not disabled

Test app:
```go
package main

import (
	"machine"
	"time"
)

func main() {
	println("starting")

	machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})

	machine.GPIO12.Configure(machine.PinConfig{Mode: machine.PinInput})
	machine.GPIO12.SetInterrupt(machine.PinRising|machine.PinFalling, handleGPIO)

	for {
		time.Sleep(1 * time.Second)
	}
}

var count = 0

func handleGPIO(pin machine.Pin) {
	count++
	machine.LED.Set(count&0x8 != 0)
}
```

Scenario: Run test app above on feather RP2040.  Apply 2KHz signal to GPIO12 using a signal generator.  Observe the frequent periods where LED is not updated (LED has irregular pulsing).  The LED should be pulsed at a consistent rate of approx 2K/8 = 250Hz

Scope output:
[image]

Yellow is the LED line
Purple is when the GC runs (I made the GC set GPIO11 high during GC runs):
```go
func runGC() (freeBytes uintptr) {
	machine.GPIO11.Configure(machine.PinConfig{Mode:machine.PinOutput})
	machine.GPIO11.High()
	if gcDebug {
		println("running collection cycle...")
	}
//...
	machine.GPIO11.Low()
	return
}
```

Observations:
1. There appears to be a 50ms pause before the GC runs (possibly low-memory condition, looking for memory blocks?)
2. It appears something is causing frequent memory allocations, possibly in the interrupt handling path, causing GC to run - probably inside the ISR context since multiple GC cycles within time.Sleep.
3. It appears interrupts are disabled during allocation and GC, and which can take 100ms+

kenbell, 2022-10-05:
It appears there is an allocation inside the GPIO interrupt handler:

> tinygo flash -target feather-rp2040 -print-allocs . -serial uart .
> /home/kenneth/work/tinygo/src/runtime/baremetal.go:42:14: object allocated on the heap: size is not constant
> /home/kenneth/work/tinygo/src/internal/task/task_stack.go:67:15: object allocated on the heap: size is not constant
> /home/kenneth/work/tinygo/src/internal/task/task_stack.go:102:12: object allocated on the heap: escapes at line 104
> **/home/kenneth/work/tinygo/src/machine/machine_rp2040_gpio.go:291:3: object allocated on the heap: escapes at line 292**
> /home/kenneth/work/tinygo/src/machine/usb.go:152:16: object allocated on the heap: escapes at line 160
> /home/kenneth/work/tinygo/src/machine/usb.go:192:17: object allocated on the heap: escapes at line 193
> /home/kenneth/work/tinygo/src/machine/usb.go:211:17: object allocated on the heap: escapes at line 212
> /home/kenneth/work/tinygo/src/machine/usb.go:116:15: object allocated on the heap: escapes at line 117
> /home/kenneth/work/tinygo/src/machine/usb.go:120:13: object allocated on the heap: escapes at line 122
> /home/kenneth/work/tinygo/src/machine/usb.go:125:13: object allocated on the heap: escapes at line 127
> /home/kenneth/work/tinygo/src/machine/machine_rp2040_usb.go:313:6: object allocated on the heap: escapes at line 324
> /home/kenneth/work/tinygo/src/machine/machine_rp2040_clocks.go:87:15: object allocated on the heap: escapes at line 87
> /home/kenneth/work/tinygo/src/machine/usb/cdc/usbcdc.go:135:8: object allocated on the heap: escapes at line 144

soypat, 2022-10-21:
@deadprogram Tag as next release? Looks extra fixed on dev tip.

> $ tinygo build -target feather-rp2040 -print-allocs . -serial uart .
/home/pato/src/tg/tinygo/src/runtime/baremetal.go:42:14: object allocated on the heap: size is not constant
/home/pato/src/tg/tinygo/src/internal/task/task_stack.go:67:15: object allocated on the heap: size is not constant
/home/pato/src/tg/tinygo/src/internal/task/task_stack.go:102:12: object allocated on the heap: escapes at line 104
/home/pato/src/tg/tinygo/src/machine/machine_rp2040_clocks.go:87:15: object allocated on the heap: escapes at line 87

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing. Thank you!

PR #3218 (2022-10-07): rp2040: remove mem allocation in GPIO ISR
Fixes #3217

It's unclear why this fixes the issue (since code is logically equivalent), but avoiding the assignment of register to a variable appears to enable some additional optimization logic to apply that prevents an 'escape'.

Have confirmed the fix by output of `tinygo flash -target feather-rp2040 -print-allocs . -serial uart .`, which no longer shows an allocation inside `machine_rp2040_gpio.go` - also confirmed on scope that test app from #3217 no longer experiences GC pauses.

I think this is a good example of why it would be good to have the [... cut]
closed 2023-02-12 by deadprogram