tinygo-org/tinygo #3225: Calling function containing defer during an interrupt handler function causes a panic

waj334, 2022-10-10:
```
var globalTestCrash testCrash

type testCrash struct {
}

func (t testCrash) test() {
	defer t.test2()
}

func (t testCrash) test2() {

}

func irqHandler(irq machine.Pin) {
	println("******************************************** ISR")

	globalTestCrash.test()
}
```

Above I have a basic interrupt handler set up that will panic once test() is called:
```
******************************************** ISR
panic: runtime error: nil pointer dereference
```

```
>tinygo version          
tinygo version 0.25.0 windows/amd64 (using go version go1.19 and LLVM version 14.0.0)
```

dgryski, 2022-10-10:
Pretty sure this is because `defer` requires an allocation, which is currently unsafe during interrupt handlers.

kenbell, 2022-10-20:
@dgryski I don't think it's an allocation issue... this test code shows no allocation, but does seem to 'not work'...

```go
package main

import (
	"machine"
	"time"
)

func main() {
	time.Sleep(2 * time.Second)

	machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})
	machine.GPIO12.Configure(machine.PinConfig{Mode: machine.PinInputPulldown})
	machine.GPIO12.SetInterrupt(machine.PinFalling, f)

	for {
		time.Sleep(1 * time.Second)
	}
}

func f(machine.Pin) {
	defer machine.LED.Set(!machine.LED.Get())
}
```

Output of `tinygo flash -monitor -target feather-rp2040 -print-allocs .`:
```
<snip>/tinygo/src/runtime/baremetal.go:42:14: object allocated on the heap: size is not constant
<snip>/tinygo/src/internal/task/task_stack.go:67:15: object allocated on the heap: size is not constant
<snip>/tinygo/src/internal/task/task_stack.go:102:12: object allocated on the heap: escapes at line 104
<snip>/tinygo/src/machine/machine_rp2040_clocks.go:87:15: object allocated on the heap: escapes at line 87
```

xudongzheng, 2024-10-14:
I'm running into this issue and it seems like the panic happens even if the function never gets deferred.

```go
package main

import (
	"machine"
	"runtime/interrupt"
)

var button = machine.D106

var i int

func test1() {
	i++
}

func test2() {
	i--
}

func main() {
	println("hello world")
	button.Configure(machine.PinConfig{Mode: machine.PinInputPullup})
	button.SetInterrupt(machine.PinToggle, func(p machine.Pin) {
		if !interrupt.In() {
			test1()
			defer test2()
		}
	})
}
```

aykevl, 2024-10-14:
We don't support defer in interrupts. We could maybe add support for it, but right now it's not supported.

Here is a PR to improve the error message: https://github.com/tinygo-org/tinygo/pull/4522
With this PR, it will say something like:

    panic: runtime error at 0x000005db: defer in interrupt

deadprogram, 2026-10-01:
Another case where this shows up: `machine/usb/msc` calls `BlockDevice.ReadAt` from the USB interrupt, so a block device whose `ReadAt` uses `defer` hangs the board as soon as the host reads from it. Filed as #5803.