tinygo-org/tinygo #3233: USB-MIDI documentation?

davidhtims, 2022-10-16:
I have a Xiao RP2040 working in that the device can be recognized as a MIDI device but I'm lost as to how to send data.
Is there any documentation for USB-MIDI?
I have asked this question in TinyGoMIDI, a Google Group I've started, and the code I have at the moment is there:
  [https://groups.google.com/g/tinygomidi/c/s8P8fqm3BaE](https://groups.google.com/g/tinygomidi/c/s8P8fqm3BaE)

sago35, 2022-10-16:
To receive and process MIDI signals from a PC or other source, the following code can be used as a reference.

https://github.com/tinygo-org/tinygo/blob/b5ad81c884981f42c1ac491cc1d195aea886ddbe/src/examples/usb-midi/main.go#L21-L25

sago35, 2022-10-16:
I think there is no documentation on USB-MIDI at this time.
But would be great to have.

davidhtims, 2022-10-17:
Thanks, just two questions please:

Why or when do I need to use the init function as mentioned in the RP2040 docs:
```
func init() {
   machine.EnableMIDI(midiTx, midiRx, midiSetup)
}
```
On trying to run the example you linked to, my installation does not know machine.BUTTON.

sago35, 2022-10-17:
You do not need to call `machine.EnableMIDI()` .
EnableMIDI is called internally.

https://github.com/tinygo-org/tinygo/blob/b5ad81c884981f42c1ac491cc1d195aea886ddbe/src/machine/usb/midi/midi.go#L21-L25

https://github.com/tinygo-org/tinygo/blob/b5ad81c884981f42c1ac491cc1d195aea886ddbe/src/machine/usb/midi/midi.go#L21-L25

So the first thing you should do is examples/usb-midi.

> I get an undeclared error on machine.BUTTON.

If `machine.BUTTON` is not defined, replace it with `machine.GPIO0` , etc.

davidhtims, 2022-10-20:
Thanks for the help, I can now send MIDI notes to the pc from Seeed Studio's Xiao RP2040.

deadprogram, 2023-09-03:
I just added PR https://github.com/tinygo-org/tinygo/pull/3889 to make some improvements to the current USB-MIDI support, please have a look.

davecheney, 2026-10-27:
Thanks for confirming you got MIDI output working. Closing as answered.

closed 2026-10-27 by davecheney