tinygo-org/tinygo #3235: SAMx51: Compiling with -opt=0 causes hardfault when disabling the FPU

waj334, 2022-10-18:
A hardfault occurs when writing to the CPACR register when compiling with -opt=0 to disable optimizations.

runtime_atsamd51.go
```
//export Reset_Handler
func main() {
	arm.SCB.CPACR.Set(0) // disable FPU if it is enabled               <------- Hardfaults occurs here
	preinit()
	run()
	exit(0)
}
```

I found that commenting out that line allows the program to run, but of course crashes eventually because FPU support is not currently present in TinyGo.

`
tinygo version 0.26.0 windows/amd64 (using go version go1.19 and LLVM version 14.0.0)
`

I am running on an ATSAME51G19A which is nearly identical to the ATSAMD51G19A variant.

waj334, 2022-11-04:
Think the FPU needs to be disabled after the call to preinit because preinit initializes the globals from the data section of the .elf. arm.SCB could have a garbage value prior to calling preinit, especially after a power cycle.

Perhaps other optimization modes causes the program to do things in a different order which causes this as it is right now to work.

aykevl, 2022-11-05:
The most reasonable way forward is probably to remove `-opt=0`. It's sometimes useful, but tends to be unreliable.
It still makes me wonder why `arm.SCB` is not a constant. It should be - but maybe that's just because that optimization is disabled with `-opt=0`.

waj334, 2023-03-04:
Turns out this issue was caused by the usage arm.AsmFull in arm.EnableInterrupts. AsmFull seemed to rely on OptimizeMaps to optimize out the map parameter supplied to it and opt=0 effectively disables this pass. As a result, the compiler was actually inserting code to also allocate memory for the hash map, which is not intended behavior! Removing the usage of arm.AsmFull seems to fix this issue.

arm.EnableInterrupts and arm.DisableInterrupts should be implemented in C using inline assembly instead.

waj334, 2023-03-04:
#3515

aykevl, 2023-05-03:
Looks like this should be fixed in the dev branch.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you!

closed 2023-06-14 by deadprogram