tinygo-org/tinygo #3241: `tinygo info bluepill` fails in ubunt 20.04 with a user installed Go SDK

dolanor, 2022-10-18:
Hi,

Using the TinyGo VS Code extension, I got a: `Could not run 'tinygo info bluepill': Error: Command failed: tinygo info bluepill requires go version 1.18 through 1.19, got go1.14` error message

I installed Go the traditional way on Ubuntu 20.04 to have it with the latest version (in `/usr/local/go`). It was Go 1.14
After that, I used the https://go.dev/doc/manage-install tutorial and mechanism to update my version of Go: it install in ~/sdk/goX.Y.Z .

TinyGo could not find this version, even though this version of `go` cmd was in my $PATH, preceding `/usr/local/go/bin`. But it kept failing giving me the info.

kenbell, 2022-10-18:
Have you tried setting GOROOT env variable?  I think this is the code that detects where Go is installed.  It has a bunch of fall-back strategies.

aykevl, 2022-10-19:
> Have you tried setting GOROOT env variable?

In my experience, this advice usually results in more problems than it solves.

I investigated this issue (long thread on Slack!) but couldn't really figure out what's going on. But it does appear that it doesn't recognize the Go SDK directory format or the way the symlinks are made, or something like that.

dolanor, 2022-10-19:
Yes, I haven't used GOROOT since I started working in Go, 8 years ago. It was already obsoleted.
I tried, though, for this issue, and in fact, it works :)

The best, I guess, would be to try `go env GOROOT` instead of `os.Getenv("GOROOT")`

davecheney, 2026-10-24:
TinyGo now obtains `GOROOT` from the `go` executable selected through `PATH`, using `go env -json GOPATH GOROOT GOVERSION`, rather than inferring the SDK directory layout. This is the approach suggested above.

You can check the selected SDK with:

```sh
go env GOROOT
tinygo env GOROOT
tinygo info bluepill
```

With TinyGo 0.44.0-dev-eae3f875 and Go 1.27.3, the first two commands report the same Go root and the Blue Pill info command succeeds.

Please open a new issue with the outputs of these commands if you encounter further SDK discovery problems.

closed 2026-10-24 by davecheney