tinygo-org/tinygo #3243: disabling usb reset for raspberry pico

alvarolm, 2022-10-20:
As I see tinygo uses parts of the official pico sdk together with other libraries to load and execute the compiled go source code, so it seems posible to implement or at least expose the underlying resources to provide useful configurations.

now I'm trying to disable usb reset, which if I were using the pico sdk, I would just set the definitions
```
PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE
PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE
```
to 0 and that would be all.

I would really appreciate an opinion o guide in any way to implement a similar feature using tinygo.

Thanks in advance

kenbell, 2022-10-20:
I don't think it's possible today - would need an enhancement.  You're looking to disable USB completely, or just the ability to reset using USB?

alvarolm, 2022-10-20:
> I don't think it's possible today - would need an enhancement.  You're looking to disable USB completely, or just the ability to reset using USB?

Understood, just the ability to reset using USB, any clues are appreciated

sago35, 2023-02-21:
The following function is used to transition to the bootloader.
https://github.com/tinygo-org/tinygo/blob/v0.27.0/src/machine/machine_rp2040_enter_bootloader.go#L50-L52