tinygo-org/tinygo #3244: support for the eZ80 and TI84 CE Plus Calculator family?

assyrianic, 2022-10-20:
I have a TI84 CE Plus calculator and it supports Python via CircuitPython. I looked up the details of the TI84 CE Plus family and saw that it uses the eZ80 family along with a coprocessor. This got me thinking if it was possible to use TinyGo on the TI84 CE Plus family. That'd be insane! Besides awesome, it'd probably be alot more efficient to use TinyGo as compared to [Circuit]Python.

aykevl, 2022-10-22:
The Z80 uses the Intel 8080 instruction set, which isn't supported by LLVM. At least not upstream, there are some forks that have it but using a LLVM fork is a lot of trouble.

So no, right now the Z80 can't be supported. Sorry. We'd first need support in LLVM itself.

adriweb, 2022-10-23:
Actually, the co-processor is an Atmel SAMD21E18, so an ARM.
The eZ80 isn't involved in _running_ Python, on the TI CE calculators.

aykevl, 2022-11-03:
Hmm, in that case it is actually feasible to do this. Without knowing the details it's hard to say how difficult this is, but at least it looks possible to do this. And it would be cool to run TinyGo on even more weird hardware (after the GBA and Switch).

I can't do this however because I don't have such a calculator.