tinygo-org/tinygo #3247: panic: cannot convert pointer value to byte when compiling with LLVM15

anuraaga, 2022-10-21:
After upgrading to the lastest `dev` and installing `llvm@15` on my Mac, a previously working build started to fail with the below exception.

```
git clone https://github.com/anuraaga/coraza-proxy-wasm.git --branch mimalloc
cd coraza-proxy-wasm
go run mage.go build
```

```
panic: cannot convert pointer value to byte

goroutine 213 [running]:
github.com/tinygo-org/tinygo/interp.rawValue.bytes(...)
	/Users/anuraag/git/tinygo/interp/memory.go:727
github.com/tinygo-org/tinygo/interp.rawValue.Uint(...)
	/Users/anuraag/git/tinygo/interp/memory.go:735
github.com/tinygo-org/tinygo/interp.(*runner).run(0x140133c2000, 0x1400a1b39f0, {0x1400549b590, 0x4, 0x1400c308360?}, 0x14011f48c90, {0x14013d94048, 0x14})
	/Users/anuraag/git/tinygo/interp/interpreter.go:866 +0x18c4
github.com/tinygo-org/tinygo/interp.(*runner).run(0x140133c2000, 0x14009f512c0, {0x1400549a090, 0x4, 0x1400aaafc50?}, 0x14011f492f0, {0x1400ce0b110, 0x10})
	/Users/anuraag/git/tinygo/interp/interpreter.go:547 +0x63ac
github.com/tinygo-org/tinygo/interp.(*runner).run(0x140133c2000, 0x14009f51270, {0x14010136830, 0x1, 0x140101648c0?}, 0x14011f48f90, {0x1400ce0a3a0, 0xc})
	/Users/anuraag/git/tinygo/interp/interpreter.go:547 +0x63ac
github.com/tinygo-org/tinygo/interp.(*runner).run(0x140133c2000, 0x14009f51220, {0x14010136650, 0x1, 0x14010958d80?}, 0x14011f48ed0, {0x1400cdf0d48, 0x8})
	/Users/anuraag/git/tinygo/interp/interpreter.go:547 +0x63ac
github.com/tinygo-org/tinygo/interp.(*runner).run(0x140133c2000, 0x14009f51130, {0x0, 0x0, 0x100ca8a68?}, 0x0, {0x10072804b, 0x4})
	/Users/anuraag/git/tinygo/interp/interpreter.go:547 +0x63ac
github.com/tinygo-org/tinygo/interp.Run({0x6000141235a0?}, 0x0?, 0x0?)
	/Users/anuraag/git/tinygo/interp/interp.go:117 +0x4e8
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x60000809a308?}, 0x140001d4a20)
	/Users/anuraag/git/tinygo/builder/build.go:1049 +0x34
github.com/tinygo-org/tinygo/builder.Build.func5(0x140081c0d20)
	/Users/anuraag/git/tinygo/builder/build.go:555 +0x49c
github.com/tinygo-org/tinygo/builder.runJob(0x140081cb380, 0x0?)
	/Users/anuraag/git/tinygo/builder/jobs.go:222 +0x48
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/anuraag/git/tinygo/builder/jobs.go:123 +0x440
Error: running "tinygo build -gc=none -opt=2 -o build/mainraw.wasm -scheduler=none -target=wasi -tags='custommalloc'" failed with exit code 2
exit status 2
```

dgryski, 2022-10-21:
Thanks.  Can you try shrinking down your program and getting a smaller reproducer?

anuraaga, 2022-10-24:
Was able to step debug and found it was just because I had built with the wrong branch, needed to be on #3245 and not on `dev` itself because the code was relying on `malloc` to not be defined by TinyGo. Sorry for the noise.

closed 2022-10-24 by anuraaga