tinygo-org/tinygo #3248: Flashing running pico timeout

soypat, 2022-10-21:
Hello, I'm running a blinky program that acquires the ADC:
```go
	machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})
	cycle := true
	for {
		machine.LED.Set(cycle)
		time.Sleep(time.Second)
		cycle = !cycle
	}
```

Trying to flash the pico while it is running fails. Around half a second after failing my computer picks the pico up. The time it takes for the pico to be picked up by my computer seems to be independent of when tinygo fails, it regularly takes around 8-10s, and tinygo seems to timeout/fail after 7 seconds.

>pato@liathink:\~/src/tg/picoboard$ tinygo flash -target=pico .
error: failed to flash /tmp/tinygo3449290731/main.uf2: unable to locate device: RPI-RP2
pato@liathink:\~/src/tg/picoboard$ tinygo flash -target=pico .
pato@liathink:\~/src/tg/picoboard$

deadprogram, 2022-11-12:
@soypat please see my PR #3275 about this very issue.

I used `-retries=25` with that branch and it consistently addresses the slowness of the rp2040 to be remounted.

soypat, 2022-11-12:
Thank you @deadprogram ! I'll close this so we may move discussion over there

closed 2022-11-12 by soypat