tinygo-org/tinygo #3254: support for multi value return

kwerner8, 2022-10-25:
Is it possible to compile an exported function in Go to WebAssembly which returns multiple values?
Go [supports](https://go.dev/tour/basics/6) the use of multiple results. So does [WebAssembly](https://webassembly.org/roadmap/).

I want to compile the following function to WebAssembly and then call it in Wasmtime:
```
//export multi
func multi = (x int32, y int32) (int32, int 32, int 32){
    return (x+y, x-y, x*y)
}
```
Currently this function compiles to:
```
(func $multi (param i32 i32 i32) 

... WebAssembly code ...)
```
It has one more input parameter and it doesn't return any values at all but it should return three values like this:
```
(func $multi (param i32 i32) (result i32 i32 i32)

... WebAssembly code ..)
```
I already had a similar problem for compiling Rust to WebAssembly. There I could enable the option _-C target-feature=+multivalue_ for the compilation and I got the right result (see here: [Multi value Wasm compilation in Rust](https://github.com/rust-lang/rust/issuethiss/73755) ). I did not find something similar in the [Using TinyGo documentation](https://tinygo.org/docs/reference/usage/). A similar problem has already been addressed in this [issue](https://github.com/tinygo-org/tinygo/issues/2512).

Is there first-class multivalue support for TinyGo yet? Or does anyone know a workaround for this issue?

aykevl, 2022-10-25:
The last time I tried I hit some issues, I don't remember exactly what it was. But at the time I decided to let the tools mature a bit more.

Note that multivalue is actually an ABI change. This means supporting it will be a backwards incompatible change. So I only want to do it when it is fully supported everywhere.

aykevl, 2022-10-25:
> It has one more input parameter and it doesn't return any values at all [...]

That extra input parameter is a pointer to the values. The ABI works as follows: values that can't be returned directly (like multiple values), have to be returned via linear memory. For callers that means they allocate some linear memory (on the C stack) and pass a pointer to the callee. The callee will store the return values at the address specified by the caller.

kwerner8, 2022-10-26:
Thank you very much for your answer!
I can access the values via the pointer to the linear memory, but I have problems with larger numbers.

As far as I know only int's up to 255 can be stored in one memory location. In the next memory location the multiples of 256 (2^8) and in the next the multiples of 2^(8*2) are stored.

With this code I can get back the int32 number in the host code:

```
int mem_addr_res = 0;
int mem_addr_res_next = 4;   // would be 8 for int64

for (int i = 0; i < FUNCTION_OUTPUT_COUNT; i++) {
    results[i].kind = WASMTIME_I32;
    long int temp_result = 0;
    for (int j = 0; j < mem_addr_res_next; j++) {
        temp_result += (wasmtime_memory_data(context, &memory)[mem_addr_start+mem_addr_res+j]) * pow(2,8*j);
    }
    results[i].of.i32 = temp_result;
    mem_addr_res += mem_addr_res_next;
}
```
The same works also for int64.

I'm having trouble passing float's with this method. Do you know how to pass float numbers with this method as a "multi-value return"?

aykevl, 2022-11-03:
I recommend you use [`DataView`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/DataView) instead of doing this manually. You can look at wasm_exec.js for examples:

https://github.com/tinygo-org/tinygo/blob/b5ad81c884981f42c1ac491cc1d195aea886ddbe/targets/wasm_exec.js#L157

Closing because I don't think there is anything to do here in TinyGo.

closed 2022-11-03 by aykevl