tinygo-org/tinygo #3258: Silent compilation failure of possibly pathological regexp on wasi

dkegel-fastly, 2022-10-29:
A large wasi app compiled fine but inexplicably started failing its smoke test when a small change was made to the source tree.

The failure had nothing obvious to do with the change; the connection only became apparent when I bisected the change.
After a little reduction, the failure was no longer silent, but involved a wasm trap.

Further bisection go the problem down to a single possibly pathological regexp, which triggers the trap all by itself.

It would be nice if that case were caught more reliably, and weren't silent in the larger test case.  Alas, I do not have a minimal reproducer for the silent part, but here is a minimal reproducer for the kind of trap that somehow because silent in the larger program:

```
$ cat foo.go
package main
import "regexp"
var r = regexp.MustCompile(`(x[a-z]*[ ]{0,10}=[ ]{0,10}y[ ]{0,10}\(.*\))`)
func main() {}
$ go run foo.go
$ tinygo run foo.go
$ tinygo run -target wasi foo.go
panic: runtime error: index out of range

Caused by:
    0: failed to invoke command default
    1: wasm trap: wasm `unreachable` instruction executed
       wasm backtrace:
           0:  0x9f0 - <unknown>!runtime.runtimePanic
           1:  0x442 - <unknown>!runtime.lookupPanic
           2: 0x3d77 - <unknown>!(regexp/syntax.patchList).patch
           3: 0x44f8 - <unknown>!(*regexp/syntax.compiler).cat
           4: 0x500f - <unknown>!(*regexp/syntax.compiler).compile
           5: 0x50d6 - <unknown>!(*regexp/syntax.compiler).compile
           6: 0x4fb4 - <unknown>!(*regexp/syntax.compiler).compile
           7: 0x50d6 - <unknown>!(*regexp/syntax.compiler).compile
           8: 0x4fb4 - <unknown>!(*regexp/syntax.compiler).compile
           9: 0x50d6 - <unknown>!(*regexp/syntax.compiler).compile
          10: 0x4fb4 - <unknown>!(*regexp/syntax.compiler).compile
          11: 0x50d6 - <unknown>!(*regexp/syntax.compiler).compile
          12: 0x4fb4 - <unknown>!(*regexp/syntax.compiler).compile
          13: 0x552c - <unknown>!(*regexp/syntax.compiler).compile
          14: 0x1e201 - <unknown>!runtime.run$1
          15: 0x1da74 - <unknown>!runtime.run$1$gowrapper
          16: 0x202ee - <unknown>!tinygo_launch
          17: 0x1d916 - <unknown>!_start
       note: using the `WASMTIME_BACKTRACE_DETAILS=1` environment variable to may show more debugging information
       
error: failed to run compiled binary /var/folders/v0/0k9hwftd29b18z9z7s2g2sfc0000gn/T/tinygo1536841757/main: exit status 134
```

dkegel-fastly, 2022-10-29:
Another clue to the silent failure: the initial trap in the larger program after I whittled it down a little looks like this:
```
2022-10-29T16:13:53.025538Z ERROR request{id=1}: WebAssembly trapped: wasm trap: uninitialized element
wasm backtrace:
    0: 0x18c279 - <unknown>!tinygo_launch
    1: 0x1fdbb - (*internal/task.Task).Resume
                    at /Users/dkegel/src/tinygo/src/internal/task/task_asyncify.go:110:17
              - runtime.scheduler
                    at /Users/dkegel/src/tinygo/src/runtime/scheduler.go:236:11
              - runtime.run
                    at /Users/dkegel/src/tinygo/src/runtime/scheduler_any.go:29:11
              - _start
                    at /Users/dkegel/src/tinygo/src/runtime/runtime_wasm_wasi.go:22:5
```

dkegel-fastly, 2022-10-29:
Happily, my app doesn't actually need the silly regexp.
But for others running into this, one possible kludge is to replace quantifiers like "{1,32}" with simpler ones like "+".

dgryski, 2022-10-29:
Nice debugging! I wonder if it's a memory corruption issue? I've had issues that led to silent corruption when the stack overflowed.

aykevl, 2022-10-30:
Interesting! I can reproduce this on the dev branch.

Unfortunately, it seems to go away with `-gc=leaking` _or_ `-scheduler=none`. Which makes debugging harder.

dkegel-fastly, 2022-10-30:
FWIW my big app uses -gc=leaking, and it was still affected...?

aykevl, 2022-10-30:
Well, that rules out an incorrectly freed object, then.

aykevl, 2022-10-30:
...could it simply be an actual stack overflow? Because I can't reproduce this anymore with `-stack-size=32kb`.

dkegel-fastly, 2022-10-30:
That's what I suspect.  I hadn't thought to try fiddling with that parameter, even though I asked for it :-)

So the bug is "stack overflows are sometimes silent", I guess, and/or "default stack on wasi is too small"?

aykevl, 2022-10-30:
> stack overflows are sometimes silent

Yeah, that's true on all platforms TinyGo supports at the moment. Not sure whether this is something we want to fix (it would be great, but I'm not sure whether there is a practical and performant way to do it).

> default stack on wasi is too small

That's subjective :) larger stacks also currently mean more data to scan each GC cycle. (This should be fixable though with a precise GC).

dkegel-fastly, 2022-10-30:
Maybe only do it on test executables...?

aykevl, 2022-10-30:
We could, but I'm not so sure that's a good idea. It could lead to rather surprising issues where a test executable works but the regular code doesn't.

dkegel-fastly, 2022-10-30:
Was any part of my problem specific to build time interpretation?  It feels like the silent failure part might have been, but I have no idea... If so, maybe build time interpretation could get a stack overflow check and/or a bigger stack.

dgryski, 2022-10-30:
The 'standard' way to check for stack overflows is with canary bytes that are overwritten and then checked at function return time.

aykevl, 2022-10-30:
> Was any part of my problem specific to build time interpretation?

I doubt it.

> If so, maybe build time interpretation could get a stack overflow check and/or a bigger stack.

That's not a thing. The interp pass is regular Go code, so it uses as much stack space as the host provides (by default 1GB).

> The 'standard' way to check for stack overflows is with canary bytes that are overwritten and then checked at function return time.

Perhaps for C (which allows arbitrarily sized stack allocations), but there are more efficient ways to do this in TinyGo because I made sure all functions have a fixed frame size. Basically, this can be done by checking the stack pointer right after incrementing it at the function entry. That's possible, but requires some cooperation with the LLVM WebAssembly backend to be efficient.

The alternative method with canary bytes seems far more involved, and requires the same amount of backend cooperation.

dkegel-fastly, 2022-11-01:
See also https://www.theregister.com/2021/11/04/webassembly_stack_canaries/

aykevl, 2022-11-02:
@dkegel-fastly there are many problems with that report, to be honest it mostly feels like FUD to me. I think the writers of that paper don't really understand the security model of WebAssembly. While it's indeed harder to work around C language deficiencies (such as memory unsafety) in the compiler, this is mostly besides the point as WebAssembly itself is sandboxed. (Possibly a hot take here)

...also, it's The Register so I'm not exactly expecting high quality journalism there.

dkegel-fastly, 2022-11-05:
Tinygo's undefined stack overflow behavior is fine as is if you're used to low level development in C or assembly, and have good tests.

It would be a bonus if it had solid stack overflow behavior at least when running on bigger targets (including wasm), but there are bigger fish to fry...

aykevl, 2022-11-05:
Yeah, it would be really useful to have. Probably as a late LLVM machine code pass. It could also be useful for Rust or for RTOSes that want to provide more safety on bare metal systems. For Linux/macOS/Windows, this can be more easily solved with guard pages (which we don't support either at the moment).

dgryski, 2022-12-06:
I wonder if we could have the function prologue check if we're about to overflow the stack and panic with a custom "stack oom" error?

aykevl, 2022-12-12:
@dgryski yeah that's what I mean, it has to be part of the LLVM backends. LLVM IR is too early in the compilation process. Or we'd need to patch binaries with a new pass like Asyncify does for example.

davecheney, 2026-10-26:
Could you retest with TinyGo 0.44-dev? I tried the regexp from your original example with current `dev` (`35f7b283`) using `-target=wasip1` and default scheduler and stack settings. It compiled during initialization and passed both matching and nonmatching checks.

If you still see the failure, please share an updated reproducer, build command, and WASM runtime and version.

dkegel-fastly, 2026-10-26:
Did we bump the default stack size?

davecheney, 2026-10-27:
I bisected this with the issue reproducer, Go 1.18.7, LLVM 14, and a 16 KiB WASI stack. This was the default stack size before and after the change, so a stack-size change did not fix it. The first passing commit is `65d65c13` (“wasm: fix GC scanning of allocas”), included in v0.27.0. Its parent still fails, but with a stack-overflow panic rather than the original index-out-of-range panic, so it is possible this has been fixed for some time.

dkegel-fastly, 2026-10-27:
Good enough for me.

closed 2026-10-27 by dkegel-fastly