tinygo-org/tinygo #3261: wasm examples and documentation update

alvarolm, 2022-10-30:
While trying to follow the guide:
https://tinygo.org/docs/guides/webassembly/
got a little confused since I couldn't figure out how to define an exported (js) function :
```
// This function is imported from JavaScript, as it doesn't define a body.
// You should define a function named 'main.add' in the WebAssembly 'env'
// module from JavaScript.
func add(x, y int) int
```
so decided to ask in slack:
[image]
https://gophers.slack.com/archives/CDJD3SUP6/p1666832165660899

thankfully @aykevl pointed out that this "method" (define an exported function) seems to be a deprecated, so the documentation needs an update https://github.com/tinygo-org/tinygo-site/issues/305, this got me wondering:

what is the correct way to call an external js function from wasm, is it using syscall/js ?
some of the wasm examples https://github.com/tinygo-org/tinygo/tree/release/src/examples/wasm seem to be over 4 years old, may be a review of them is necessary ?

glad to help in any way

deadprogram, 2023-05-17:
Please see https://github.com/tinygo-org/tinygo-site/pull/349 and https://github.com/tinygo-org/tinygo-site/pull/352

I will close this issue on next release when these updates are published. Thank everyone!

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you!

closed 2023-06-14 by deadprogram