tinygo-org/tinygo #3274: Setuid not declared by package syscall

tachyonicbytes, 2022-11-06:
Hello! I am trying to port [gau](git@github.com:lc/gau.git) to wasm, using `tinygo`.

The package compiles fine to wasm with `GOOS=js GOARCH=wasm`, but I want to use `tinygo`, in order to get better `wasi` bindings and be able to use it as is with more wasm runtimes.

The package uses some libraries that are not yet usable with tinygo ([fasthttp](github.com/valyala/fasthttp) and [json-iterator-go](github.com/json-iterator/go)), but I was able to swap them for `tinygo/drivers/net/http` and `encoding/json` in my [fork](https://github.com/tachyonicbytes/gau), so the compilation problems went away. The code still compiles and works with go, but I ran into `tinygo -target=wasi` compilation problems that I was not able to resolve.

The main problem now stands at:
```
tinygo build -o ./build/gau -target=wasi ./cmd/gau/main.go 
# golang.org/x/sys/unix
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_linux.go:1896:17: Setuid not declared by package syscall
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_linux.go:1900:17: Setgid not declared by package syscall
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_linux.go:1904:17: Setreuid not declared by package syscall
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_linux.go:1908:17: Setregid not declared by package syscall
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_linux.go:1912:17: Setresuid not declared by package syscall
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_linux.go:1916:17: Setresgid not declared by package syscall
go/pkg/mod/golang.org/x/sys@v0.1.0/unix/syscall_unix.go:524:17: Exec not declared by package syscall
```

As the code does not use the syscall package directly, I am not sure exactly which is the problematic call, and how to solve it. I looked around in the tinygo github issues and it seems like these kinds of issues have some chance of being solved (https://github.com/tinygo-org/tinygo/issues/1286).

Is there anyway to make it work for now or it cannot be solved until tinygo supports more of golang? The main idea is that since tinygo/drivers/net/http is supported, and the package itself is basically a http client plus some processing and reporting, I am also not sure exactly which library uses the syscall package. Is there a way to identify this, barring looking into each package itself?

dgryski, 2022-11-07:
This should be easy to stub given that Setuid isn't supported in Go and just returns an error.

https://github.com/golang/go/issues/1435

tachyonicbytes, 2022-11-08:
@dgryski The stubbing should be done at `tinygo` level, right? I found this [github issue](https://github.com/tinygo-org/tinygo/pull/2574) that seems to be very similar to this problem.

I could not find a file named `syscall_unix.go` in `src/syscall/`, so I don't really know where I could add the stubs.

Is there something I can do at my project level? Would adding a file that has `package syscall` at the top and implements the stubs be enough to continue compiling (until I hit the next error, or until everything compiles successfully)?

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing. Thanks!

closed 2023-02-12 by deadprogram