tinygo-org/tinygo #3277: wasm heap corruption when pointers are in wasm locals

anuraaga, 2022-11-08:
While understanding how wasm locals get marked during GC, I found this case where they don't seem to get marked causing heap corruption of valid objects.

```
git clone https://github.com/anuraaga/tinygo-gctest
cd tinygo-gctest
tinygo build -opt=2 -scheduler=none -target=wasi -o gctest.wasm .
go run github.com/tetratelabs/wazero/cmd/wazero@latest run gctest.wasm
```

Should print yogi and garfield, but prints garfield twice because yogi got overwritten

anuraaga, 2022-11-08:
BTW updated my repro command to compile without `-opt=2`, verified it shows the same behavior

anuraaga, 2022-11-08:
I found that reverting #3230 (with some yolo of me updating to llvm15 world - https://github.com/tinygo-org/tinygo/compare/dev...anuraaga:revert-3230?expand=1) seems to solve the issue for this code. Not a root cause but hopefully can be a good hint.

dgryski, 2022-11-08:
@anuraaga So do you think the issue was related to the alloca scanning patch or the llvm15 upgrade?

anuraaga, 2022-11-08:
@dgryski I think the allocas scanning patch but the llvm upgrade is also a significant one that could have had an effect, not sure.

dgryski, 2022-11-14:
I rebuilt tinygo with my homebrew llvm (15.0.4) and this test case stopped failing for me.  I agree though I'd be happier knowing the actual root cause.

dgryski, 2022-11-15:
Here's the diff of the body of the for loop, from the `scheduler=none` to the regular wasm scheduler:
```
 for.body:                                         ; preds = %for.loop
-  %3 = tail call fastcc i32 @runtime.runGC(), !dbg !2116
-  %4 = tail call fastcc ptr @main.newCat(), !dbg !2119
-  %5 = add nuw nsw i32 %2, 1, !dbg !2120
-  call void @llvm.dbg.value(metadata i32 %5, metadata !2105, metadata !DIExpression()), !dbg !2120
-  br label %for.loop, !dbg !2113
+  %7 = tail call fastcc i32 @runtime.runGC(), !dbg !2400
+  %8 = tail call fastcc ptr @main.newCat(), !dbg !2403
+  %9 = getelementptr { ptr, i32, ptr, ptr, ptr, ptr, ptr }, ptr %gc.stackobject, i32 0, i32 4, !dbg !2404
+  store ptr %8, ptr %9, align 4, !dbg !2404
+  %10 = add nuw nsw i32 %6, 1, !dbg !2404
+  call void @llvm.dbg.value(metadata i32 %10, metadata !2389, metadata !DIExpression()), !dbg !2404
+  br label %for.loop, !dbg !2397
```

We can see that for `scheduler=none`, the result of the call to `newCat()` (`%4`) is ignored, whereas with the default scheduler, the result (`%8`) is stored in a stack object (`+  store ptr %8, ptr %9, align 4, !dbg !2404`).

dgryski, 2022-11-16:
The last read from `stackChainStart` was removed in https://github.com/tinygo-org/tinygo/pull/3230/files#diff-ccb1c1473c763d58f3a7b51510933007b70dbcd799025b5ae5e06237e0285afaL27 , and with `scheduler=none` the llvm optimizer sees that writes to `stackChainStart` are dead stores, and so writes to the `gc.stackobject` are also dead stores, and so removes all the pointer tracking writes.  Thus, we lose the pointer.

aykevl, 2022-11-17:
@anuraaga can you test https://github.com/tinygo-org/tinygo/pull/3291? It fixes the bug for me in the reproducer, but it would be useful if you could verify the bug fix in the original code you derived this reproducer from.

Also, just wanted to say thanks for the very small reproducer. It really helps!

anuraaga, 2022-11-18:
Thanks @aykevl - seems to work fine https://github.com/corazawaf/coraza-proxy-wasm/pull/93

Might want to add something similar to the reproducer as a unit test to prevent regression in the future.

closed 2022-11-18 by anuraaga